ArduPilot/MissionPlanner #1039: Request: Automatic logbook feature

iskess, 2015-10-20:
The insurance companies are asking how much UAV flight time the operator has. The FAA could also want a logbook in the upcoming regulation. It seems like it would be a simple matter to create an automatic logbook in Mission Planner.
I imagine a simple .xls or .csv file that gets written into by MP. The log would only start when the aircraft is armed. SO make it more robust, it could require some other verification that the aircraft is flying. The logged data would be user defined, but a basic default would be:
- Date, 
- Time, 
- Aircraft (sysID), 
- Location (it would be nice to assign a name to a radius around a lat/long), or perhaps Google maps can use the nearest location name. 
- Duration of flight,
- Associated tlog name
- Notes or comments
Other custom data fields for, # of WP's, mAh battery used, etc
There should also be a threshold filter that eliminates logs with a duration shorter than X minutes.

meee1, 2015-10-20:
under normal use? do you delete your tlogs?
have a look at "logindex" under control-f. as this is what I would most likerly expand out.

iskess, 2015-10-21:
To answer your question, I usually try to delete the rlogs, and useless tlogs to keep the folder full of only relevant data. 
I hadn't seen logindex before, nice! I love that it incorporates the thumbnail. I wish I could delete useless logs directly from that tool, just by glancing at the flight path (of lack thereof). If a single click could delete the .tlog, .rlog, and .jpg, that would be very useful. Wouldn't it be nice to be able to associate a dataflash log too for future reference, even if done manually?
Logindex could be the basis for a great logging tool. It needs to write to an editable and savable file type that can be amended to. 
Thank you for your consideration.

gediminasgu, 2019-03-24:
#2107 Has a possibility to delete unuseful logs.