ArduPilot/MissionPlanner #1041: Auto Analysis doesn't work with 3.3

andersonpaac, 2015-10-21:
Shows bad input file for all the .bin/.log files. The log file is located at

https://drive.google.com/file/d/0B3HJB33JQkn2RFJ4SFBzNWpGX28/view?usp=sharing

rmackay9, 2015-10-21:
I'm pretty sure that Peter Barker's work on fixing up the Log Analyser will largely resolve this so I think a good step forward would be for @meee1 to try updating the MP's analyzer to the latest version.

peterbarker, 2015-10-21:
I ran this through LogAnalyzer.py from diydrones/master and it processed your log just fine.

Older versions of LogAnalyzer.py would certainly have choked.  Once @meee1 gets a chance to update it, you should be all good.

```
pbarker@bluebottle:~/rc/ardupilot/Tools/LogAnalyzer(log-download-and-rotation-fixes)$ ./LogAnalyzer.py ~/2015-10-20\ 11-12-39.log
Dataflash log analysis report for file: /home/pbarker/2015-10-20 11-12-39.log
Log size: 17.03mb (215637 lines)
Log duration: 0:36:39

Vehicle Type: ArduCopter
Firmware Version: V3.3 (340970fc)
Hardware: 
Free RAM: 0

Test Results:
                   GPS:  FAIL       Min satellites: 4, Max HDop: 42.51                         [GRAPH]
                   VCC:  GOOD                                                              
         Dupe Log Data:  GOOD                                                              
        Event/Failsafe:  FAIL       ERR found: FLT_MODE                                        [GRAPH]
                 Empty:  GOOD                                                              
               Compass:  FAIL       Large change in mag_field (36.07%)                         [GRAPH]
                                    Max mag field length (555.70) > recommended (550.00)
              Brownout:  GOOD                                                              
            Parameters:  GOOD                                                              
          IMU Mismatch:  WARN       Check vibration or accelerometer calibration. (Mismatch: 1.16, WARN: 0.75, FAIL: 1.50)    [GRAPH]

The Log Analyzer is currently BETA code.
For any support or feedback on the log analyzer please email Andrew Chapman (amchapman@gmail.com)

pbarker@bluebottle:~/rc/ardupilot/Tools/LogAnalyzer(log-download-and-rotation-fixes)$
`
```

I also ran it through a pre-release version of dronekit-la.  Here's a couple of worrying snippets for you:

```
    "Arming Checks" : {
         "description" : "This test will FAIL if the craft ever arms when some arming checks are disabled",
         "evilness" : 132,
         "name" : "Arming Checks",
         "results" : [
            {
               "evidence" : [
                  "Arming flags: 0",
                  "NONE check disabled",
                  "ALL check disabled",
                  "BARO check disabled",
                  "COMPASS check disabled",
                  "GPS check disabled",
                  "INS check disabled",
                  "PARAMETERS check disabled",
                  "RC check disabled",
                  "VOLTAGE check disabled",
                  "BATTERY check disabled",
                  "AIRSPEED check disabled",
                  "LOGGING check disabled"
               ],
               "evilness" : 22,
               "reason" : "Some of the arming checks were disabled when the craft was  armed",
               "series" : [ "PARM.Name", "PARM.Value", "EV.Id" ],
               "severity-score" : 22,
               "status" : "FAIL",
               "timestamp" : 48425738

---

{
               "duration" : 0.5075460076332092,
               "duration-units" : "seconds",
               "evidence" : [
                  "threshold=10.000000",
                  "threshold-duration=0.250000 seconds",
                  "Max-Delta=28.000000 degrees",
                  "Roll-at-Max-Delta=65.000000 degrees",
                  "Desired-Roll-at-Max-Delta=37.000000 degrees",
                  "Pitch-at-Max-Delta=-6.000000 degrees",
                  "Desired-Pitch-at-Max-Delta=16.000000 degrees",
                  "Motor 1 clipping high"
               ],
               "evilness" : 10,
               "reason" : "Desired attitude not achieved",
               "series" : [
                  "ATT.Roll",
                  "ATT.Pitch",
                  "ATT.Yaw",
                  "ATT.DesRoll",
                  "ATT.DesPitch",
                  "ATT.DesYaw"
               ],
               "severity-score" : 10,
               "status" : "FAIL",
               "timestamp_start" : 992076544,
               "timestamp_stop" : 992584090
            },
            {
````

Your motor1 might be dying...

meee1, 2015-10-21:
mp is updated here https://github.com/diydrones/MissionPlanner/commit/fae2a5297dcd74ec7064c31f1c3062cbcfab3f71

andersonpaac, 2015-10-22:
@peterbarker 

Could you explain more on the motor1 clipping?. I see the AP is having issues having it to achieve the required attitude.

closed 2015-10-22 by andersonpaac