ArduPilot/MissionPlanner #1044: Add support for hash-checked local device parameter cache

NaterGator, 2015-10-23:
This is the corresponding groundstation feature for [ardupilot parameter hash](https://github.com/diydrones/ardupilot/issues/3070), implemented on QGroundControl [here](https://github.com/mavlink/qgroundcontrol/pull/1965) for PX4.

The premise is to save a local copy of connected device parameters (ideally in a table based on device ID) that is checked when a new device connects. If a locally saved copy of parameters exists MP should perform a CRC32 hash over the parameter table and request the value of the magic parameter _HASH_CHECK from the connected device. From that request there are three possible outcomes:
  1. The device responds with a CRC32 hash that matches the locally stored copy of parameters; the local copy is used in place of requesting the full list of parameters+values from the remote device and connect initialization continues quickly
  2. The device responds with a negative acknowledge or nonmatching CRC32 hash, the locally cached parameters are thrown out, and the normal PARAM_REQUEST_LIST message is used to retrieve a copy of all params+values from the remote device as normal.
  3. The device never responds to the _HASH_VALUE parameter request and a timer expires, triggering the normal  PARAM_REQUEST_LIST message to retrieve a copy of all params+values from the remote device as normal.

The benefit is obviously that most of the time parameters are not expected to change and we can skip the bandwidth and time-intensive process of syncing everything over the data link at initial connect.

meee1, 2015-10-23:
things like the baro pressure and gyro calibration numbers are going to change on each reboot. so it will only work with reconnecting to an existing session

NaterGator, 2015-10-23:
Are those values written to EEPROM? Is there a notion of volatile and
nonvolatile parameters?