ArduPilot/MissionPlanner #1049: Serial Inject GPS addresses null Pointer

denniszollo, 2015-10-30:
Serial Inject GPS fails with this error when using SwiftNav Piksi.  Seems like a Null pointer exception that may be related to a buffer overrun.

33:05,458 ERROR MissionPlanner.SerialInjectGPS - System.NullReferenceException: Object reference not set to an instance of an object.
   at MissionPlanner.Comms.UdpSerial.get_BytesToRead() in c:\Users\michael\Source\Repos\MissionPlanner\ExtLibs\Comms\CommsUdpSerial.cs:line 57
   at MissionPlanner.SerialInjectGPS.mainloop() in c:\Users\michael\Source\Repos\MissionPlanner\SerialInjectGPS.cs:line 125 (c:\Users\michael\Source\Repos\MissionPlanner\SerialInjectGPS.cs:136) [injectgps]

denniszollo, 2015-10-30:
One entire log where the issue occurred available here.

meee1, 2015-10-30:
what is you testing enviroment?

ie I assume piksi console? using the udp?

can you provide some more info so I can replicate.

denniszollo, 2015-10-30:
The setup basically follows the WIKI page I wrote, which worked as of July and is availble here: http://docs.swiftnav.com/wiki/Integrating_Piksi_with_the_Pixhawk_platform

Here are the cliffnotes:
Intel MAC running windows natively
Piksi Console forwarding observations over UDP
Mission planner set up as UDP host
Connect to Pixhawk over USB in Mission Planner
Piksi GPS device attached to Pixhawk over Serial 4/5 @115200 baud

Do you require a simulator of the UDP traffic?  I can spin one up for you.

meee1, 2015-10-30:
dennis, could you try MP beta?
looks like it might have been a udp circle buffer bug.

denniszollo, 2015-10-30:
MP Beta seems to have done the trick, It has run continuously for many minutes.

closed 2015-10-30 by meee1