ArduPilot/MissionPlanner #1053: Add CHX_OPTS for 9 to 12

hamishwillee, 2015-11-05:
From Copter 3.3 you can set CH_9_OPTS - CH12_OPTS as well as CH7_OPTS and CH8_OPTS. Would be good to add the options to the parameters in the full parameter lists and also add the UI options in: Config/Tuning | Extended Tuning

rmackay9, 2015-11-06:
the parameter list should already have the new descriptions but as written above, adding new drop-downs to the extended tuning screen would be nice.

hamishwillee, 2015-11-06:
@rmackay9 The parameter descriptions for  CH_9_OPTS - CH12_OPTS weren't showing up in the Mission Planner "Full Parameter List" on Copter 3.3.0 when I tested it earlier (SITL)

rmackay9, 2015-11-06:
@hamishwillee, that will be the age-old problem of the MP on your machine having out of date parameter descriptions.  In mine they show up.

hamishwillee, 2015-12-21:
@meee1 Did you note the original request for addition of CH_9_OPTS - CH12_OPTS in the **Config/Tuning | Extended Tuning** screen?

meee1, 2015-12-22:
oh nope, I assumed it was good from randys request. ill reopen

lvale, 2016-01-24:
@meee1 Would be helpful to also have at least 12 channels displayed on the Radio Calibration screen. People don't know they can use channels greater than 8, because of this.

hamishwillee, 2016-01-25:
@lvale Suggest you raise a different request for that :-)

lvale, 2016-01-25:
@hamishwillee thanks. done.