ArduPilot/MissionPlanner #1054: Current Beta Issues

Jman841, 2015-11-06:
1. In the current beta mission planner (.10) the "Change Alt" and "Change Speed" buttons are in Chinese.

2. Elevation graph still does not work reliably. Sometimes It works, sometimes I get an error message. It was much more stable before in the stable release as it never gave me an error in the past.

3. The prefetch map data no longer goes beyond the zoom level you start at. So if you prefetch for a certain area at zoom level 14, it will download the maps up to zoom level 14 but not beyond that so you can't zoom in any further.

4. User items on the HUD do not work after the inital time. For instance, I set "timeinair:" to show me the time in air and it works fine when I first enable it. However, if I close mission planner and disconnect the plane, then fly again, it will just read 0. This is the same issue for other items I have tested. If I go back to "User Items" then uncheck the parameter and re-enable it, it will work fine again.

5. Preflight checklist is not editable and only set up for planes.

Thank you!

meee1, 2015-11-15:
1. fixed 7c0f595a52e2021180a7a00a1095db2347d98656
2. can you provide the error you receive?
3. not aware of this one, it may be a google issue
4. ill check out
5. preflight was just for testing, it wont be in a release build until its rewritten

meee1, 2015-11-15:
3. fixed 987cbb4c

meee1, 2015-11-15:
with 4. can I get more detail on how you operate?
ie steps to reproduce. including what your connecting to etc

Jman841, 2015-11-28:
Sorry for the Delay, I was traveling and crazy bus.

2. I will save the error report next time it happens.
4. It seems to be hit or miss. I have the flight time and mah used shown in the HUD and sometimes it works perfect an other times it just reads 0 the entire time. However, if I remove the data then put it back into the HUD it works everytime. Next time it happens i'll record what ardupilot firmware and the procedures that happened.
5. ok cool, I like the preflight a lot!

gulp79, 2016-02-29:
In current beta, the "GoogleHybridMap" is black, i can see roads and names, but no satellite images.