ArduPilot/MissionPlanner #1055: Moving base not working

pepevalbe, 2015-11-11:
I would like to use my interanl laptop's GPS in mission planner.
I haven't used it yet but **Follow me** seems to be working:
[image]

However **MovingBase** feature seems to be broken. I can't see any NMEA sentece and home positition does not change.
[image]

meee1, 2015-11-15:
the code between these are identical.

can you please post these files
MovingBase.txt

followmeraw.txt

from the mp directory

pepevalbe, 2015-11-15:
Here are the files:
[followmeraw.txt](https://github.com/diydrones/MissionPlanner/files/34851/followmeraw.txt)
[MovingBase.txt](https://github.com/diydrones/MissionPlanner/files/34852/MovingBase.txt)

They seem to be ok.

henrik04, 2016-07-04:
I got the moving base "window" working, updating the numbers and updating the "flight window" little purple icon (looking like a way point but blue/purple). 
So all this looks good in Mission Planner 1.3.38 Only problem is that RTL (return to launch) ignore completely this "moving base" {Arducopter 3.3.3). I tried to  enable the [update rally point 0] tick box. makes absolutely no difference, unfortunately. 
Henri

aticof, 2019-05-23:
Hi henrik04, I have the same problem as you, did you finally get a correct RTL?