ArduPilot/MissionPlanner #1061: Unable to download dataflash logs via mavlink with Plane 3.4.0 and Pixhawk

WickedShell, 2015-11-19:
Others are able to download logs with the same version (Tridge tested against the same versions). The pixhawk's first log is 248 last is 317. mavproxy succesfully lists all the logs, while mission planner 1.3.32 throws an alert of Error receiving log list.

meee1, 2015-11-20:
could you please upload missionplanner.log?
just want to see where the error is occuring

WickedShell, 2015-11-20:
I believe this is the appropriate log

http://files.swiftradioplanes.com/MissionPlanner.log

meee1, 2015-11-20:
did log 253 exist?

2015-11-20 14:21:35,192  INFO MissionPlanner.MAVLinkInterface - id 248 lastllogno 323 #logs 76 size 88662016 (c:\Users\michael\Source\Repos\MissionPlanner\Mavlink\MAVLinkInterface.cs:3173) [Base Thread]
2015-11-20 14:21:35,192  INFO MissionPlanner.MAVLinkInterface - GetLogEntry 248-248 (c:\Users\michael\Source\Repos\MissionPlanner\Mavlink\MAVLinkInterface.cs:3206) [Base Thread]
2015-11-20 14:21:35,929  INFO MissionPlanner.MAVLinkInterface - GetLogEntry 249-249 (c:\Users\michael\Source\Repos\MissionPlanner\Mavlink\MAVLinkInterface.cs:3206) [Base Thread]
2015-11-20 14:21:36,734  INFO MissionPlanner.MAVLinkInterface - GetLogList Retry 3 - giv com True (c:\Users\michael\Source\Repos\MissionPlanner\Mavlink\MAVLinkInterface.cs:3220) [Base Thread]
2015-11-20 14:21:37,460  INFO MissionPlanner.MAVLinkInterface - GetLogEntry 250-250 (c:\Users\michael\Source\Repos\MissionPlanner\Mavlink\MAVLinkInterface.cs:3206) [Base Thread]
2015-11-20 14:21:38,180  INFO MissionPlanner.MAVLinkInterface - GetLogEntry 251-251 (c:\Users\michael\Source\Repos\MissionPlanner\Mavlink\MAVLinkInterface.cs:3206) [Base Thread]
2015-11-20 14:21:38,897  INFO MissionPlanner.MAVLinkInterface - GetLogEntry 252-252 (c:\Users\michael\Source\Repos\MissionPlanner\Mavlink\MAVLinkInterface.cs:3206) [Base Thread]
2015-11-20 14:21:39,626  INFO MissionPlanner.MAVLinkInterface - GetLogEntry 253-253 (c:\Users\michael\Source\Repos\MissionPlanner\Mavlink\MAVLinkInterface.cs:3206) [Base Thread]
2015-11-20 14:21:40,368  INFO MissionPlanner.MAVLinkInterface - GetLogList Retry 3 - giv com True (c:\Users\michael\Source\Repos\MissionPlanner\Mavlink\MAVLinkInterface.cs:3220) [Base Thread]
2015-11-20 14:21:41,199  INFO MissionPlanner.MAVLinkInterface - GetLogList Retry 2 - giv com True (c:\Users\michael\Source\Repos\MissionPlanner\Mavlink\MAVLinkInterface.cs:3220) [Base Thread]
2015-11-20 14:21:42,023  INFO MissionPlanner.MAVLinkInterface - GetLogList Retry 1 - giv com True (c:\Users\michael\Source\Repos\MissionPlanner\Mavlink\MAVLinkInterface.cs:3220) [Base Thread]

WickedShell, 2015-11-20:
Yup, 19,460,096 bytes, converts correctly with no problem.

WickedShell, 2015-11-20:
When I look at a different log/try again it fails on a different number. Doesn't seem to be specific to 253.

meee1, 2015-11-21:
could I get the tlog as well?

meee1, 2015-11-21:
it will be large, as it will have the df log inside it.

WickedShell, 2015-11-21:
I believe this an appropriate .tlog. (I have far to many of them in the folder).

http://files.swiftradioplanes.com/2015-11-20%2014-21-26.tlog

meee1, 2015-11-21:
2015-11-21T05:21:39.626	FE	 E	49	1	1	76	mavlink_log_entry_t	time_utc
2015-11-21T05:21:39.626	FE	6	25	FF	BE	75	mavlink_log_request_list_t	start
2015-11-21T05:21:40.369	FE	6	27	FF	BE	75	mavlink_log_request_list_t	start
2015-11-21T05:21:41.199	FE	6	29	FF	BE	75	mavlink_log_request_list_t	start
2015-11-21T05:21:42.024	FE	6	2A	FF	BE	75	mavlink_log_request_list_t	start
2015-11-21T05:21:43.064	FE	 E	6E	1	1	76	mavlink_log_entry_t	time_utc

from what I can see mp did what it was meant to.

but for some reason it took the autopilot 4 seconds to response to a request.
I will be increasing the timeout from 700ms to 1000ms and retry count from 3 to 5. this should resolve this specific case.

closed 2015-11-24 by meee1