ArduPilot/MissionPlanner #1067: Can't build installer create.bat fails

kjeremy, 2015-11-24:
I'm having trouble building the installer.  I run `create.bat` and get:

```
installer.wxs(121) : error LGHT0130 : The primary key `ngen_MissionPlannerexe` is duplicated in table `NetFxNativeImage`.  Please remove one of the entries or rename a part of the primary key to avoid the collision.
```

Then it continues on, creates `MissionPlanner-1.3.32.zip` but then creates two 1KB files: `MissionPlanner-latest.zip` and `MissionPlanner-latest.msi`

kjeremy, 2015-11-24:
I think the error above is stopping wix from building `MissionPlanner-1.3.32.msi`.

meee1, 2015-11-24:
works for me just fine
using
Windows Installer XML Toolset Linker version 3.9.1208.0

meee1, 2015-11-24:
the 2 1kb files are also intentional, they are symlinks.

the entire wix project is really just for my deployment.

closed 2015-11-24 by meee1