ArduPilot/MissionPlanner #1070: Digicam feature

jcooper21, 2015-11-28:
It would be nice if the digicam feature with engine cut was easier to setup. When planning a mission lots of the settings have to be manually changed and this sticks when dealing with lots of waypoints.

meee1, 2015-12-16:
I'm not even sure on how this is setup now, can you please show me your references.

jcooper21, 2015-12-16:
When you select digicam configure in the waypoint window there is the engine cut option. I can't find any documentation on the engine cut feature. I assume we have to configure each waypoint with a digicam configure, digicam control and a waypoint for each picture location. The workflow is not very simple.

meee1, 2015-12-21:
currently as it stands, the autopilot has no feedback to even override the throttle. it is just passed on to other mavlink devices, so at the point in time I will be removing that reference

closed 2015-12-21 by meee1