ArduPilot/MissionPlanner #1071: Survey grid screen

jcooper21, 2015-11-28:
It would be nice if the survey grid screen was able to be seen after the mission is planned and accepted for reference purposes. It could be as simple as a button on the right side of the mission planning screen under the write data button.

kikislater, 2015-11-28:
+1
And also export it to kml/shape

meee1, 2015-12-16:
what is the intent behind this request. just trying to understand the reason, and to see if there are better alternatives.

iskess, 2015-12-16:
I believe the intent is to be able to save and edit a survey grid. If for example you want to make a small change after hitting Accept. Or if you want to fly the same grid as last week, but change the camera.

meee1, 2015-12-16:
try control-s and control-o
s to save
o to load

while in the grid screen

iskess, 2015-12-16:
That works!
Thanks. 

Iam

meee1, 2015-12-21:
I'm going to close this, as ive partially answered ad have received no further response

closed 2015-12-21 by meee1