ArduPilot/MissionPlanner #1072: Updated Heli Setup Page

R-Lefebvre, 2015-11-30:
Please update the helicopter setup page for AC3.3 according to this marked up drawing.  Any questions, please ask.

[image]

meee1, 2015-12-08:
Rob, how does
 // @Values: 0:Disabled,1:Passthrough,2:Max collective,3:Mid collective,4:Min collective

translate to the buttons?

R-Lefebvre, 2015-12-08:
Same order.  I had to try to come up with language that was simple to understand for people using the setup page.

0: Disabled, Active
1: Passthrouth, Manual
2: Max collective, Max
3: Mid collective, Zero
4: Min collective, Min
5: Oscillate(need to fix this), Test

meee1, 2015-12-16:
this should be in the current beta for the most part

R-Lefebvre, 2015-12-24:
Checked it out, looks really good.  Here are a few other things I noticed:

[image]

When the user enters manual servo mode, MP appears to still be writing values into the Min and Max Collective and HS4 parameters depending on how far they move the sticks.  This should no longer be happening.  The intent is that these fields are now full manual.  The only time that MP should be writing value into them is when the user types in a value.