ArduPilot/MissionPlanner #1073: GPS Status = 4

jeremylindsell, 2015-12-02:
If GPS Status is recorded as 4 in the data flash log (no idea why this happens, but it does) then Mission planner has the following problems:
1. After MAVLINK download, MP names the log files after the time/date of download rather than the time/date of the mission.
2. When reviewing the data flash log in MP, the date and times show as 0001-01-01 00:00:00.000 throughout the file
3. Geotagging fails with an "out of range" error.
This is only problematic if GPS Status = 4 applies throughout the mission. If status starts as 3 (and presumably lower, not checked) and changes to 4 in mid flight then MP seems to cope with that OK. Editing the log file to change the GPS Status = 4 to 3 corrects problems 2 and 3.

jeremylindsell, 2015-12-02:
Confirmed that if the first GPS message in the dataflash log has GPS Status = 4 MP can't work out all the subsequent date/times from the counters. Changing just the first GPS Status value to 3 allows MP to work out the dates properly even if all subsequent GPS status values are 4.

Also, I'm glad to have learned that 4 is a legit GPS status meaning "3D with a good HDOP"

So this is a little bug I think.

Thanks

Jeremy

meee1, 2015-12-02:
could you please try beta MP
help > update beta.
and report back?

WickedShell, 2015-12-03:
@jeremylindsell GPS Status 4 is valid DGPS. Typically what that means is 3D fix with SBAS.

jeremylindsell, 2015-12-03:
Thanks @meee1 I will let you know.
Thanks @WickedShell the flight was in Ghana which lacks SBAS so that's intriguing.

WickedShell, 2015-12-03:
@jeremylindsell I'm assuming that was with APM and a ublox GPS?

jeremylindsell, 2015-12-03:
@WickedShell yes, sorry I should have included all that: Arducopter 3.3 on Pixhawk with UBLOX Neo 8 gps/compass (well it was sold as that but I don't know how to confirm that it really is) and Mission Planner 1.3.32.

WickedShell, 2015-12-03:
So you're GPS thinks it found a valid SBAS source that it could use. My
guess would be that you accidentally locked EGNOS (You're outside the
intended coverage area but you it's not a stretch to think that you might
be receiving it. You could disable SBAS on the GPS and then you shouldn't
see a GPS status of 4 anymore. If you're Ublox 8 is a capable of doing RAW
data we can determine exactly which SBAS it found from the GPS.

jeremylindsell, 2015-12-03:
Interesting. Hopefully beta MP will resolve the status issue - can't try it just yet as computer is grinding away on something else. 

And if my gps is getting SBAS then I'd rather take advantage of that. Also, I don't know how to interact with my gps to do things like disable SBAS or look for RAW data unless it can be done in Mission Planner via Pixhawk. Where would you expect any RAW data to be stored? In the BIN file from the flight or on board the GPS? I'm not in Ghana now so can't switch on and reproduce the conditions of that flight.

WickedShell, 2015-12-03:
I don't expect any raw data unless you explicitly enabled it and bought
hardware for the purpose :) (It can be done on the param list/param tree
with GPS_RAW_DATA, but if you didn't do it on purpose then its a fairly
safe assumption that you don't have it. RAW would be in the AP log if it
was recording.

jeremylindsell, 2015-12-11:
Didn't get a chance to try and beta version before the new MP update came along and that seems to work fine now with GPS Status=4 in line 1. Many thanks.

closed 2015-12-11 by jeremylindsell