ArduPilot/MissionPlanner #1075: DF Bin -> Log then Revieing that .log causes null exception

magicrub, 2015-12-10:
v1.3.33 build 1.1.5820.11240 regression.

"Review a Log" using any .bin works
"Review a Log" using a .log generated from <=v1.3.32 works
Pressing "PX4 .Bin to .Log" and then reviewing that .log causes a nullExcepion

Reports are that it is broken on Rover and Plane but works for Copter logs.

meee1, 2015-12-15:
tom, I have tested all the firmwares, and cant replicate this issue.

meee1, 2015-12-16:
can you provide the specific log

magicrub, 2015-12-16:
I just tried it on another new log from master and it worked on v1.3.33 but
still fails on this log from master a month ago
https://drive.google.com/open?id=0B6RPB3yIS_0MRGNhMzVqbklmLWM

magicrub, 2015-12-16:
That is, ardupilot master

meee1, 2015-12-16:
the issue is this

MSG, Demo Servos!
MSG, Demo Servos!
MSG, Initialising APM...
MSG, Demo Servos!
MSG, 

 Ready to FLY.
STRT, 7, 41

there are extra line feeds for some reason.

meee1, 2015-12-16:
this should be resolved.

closed 2015-12-16 by meee1