ArduPilot/MissionPlanner #1084: Elevation profile Graph crashes

jmachuca77, 2015-12-15:
When trying to display the elevation graph I get a crash with the following error:
[image]

meee1, 2015-12-15:
I think its something specific in your mission that's causing this. but ive added a null check anyway.

jmachuca77, 2015-12-15:
I have tried on several missions and I get the same result. all the missions were done with the survey grid . I'll upload my mission if it helps.

jmachuca77, 2015-12-15:
This is the mission:

[elevationgraphcrash.waypoints.txt](https://github.com/diydrones/MissionPlanner/files/63281/elevationgraphcrash.waypoints.txt)

meee1, 2015-12-15:
thanks for the wp file, that shoed me the issue. next beta will be fixed.

closed 2015-12-16 by meee1