ArduPilot/MissionPlanner #1085: Throwing an error in UI when selecting "Camera Gimbal" in Optional Hardware config

travmason, 2015-12-17:
In the UI when I'm selecting the Camera Gimbal config in the Initial Config/Optional Hardware menu it is throwing the attached error.

Also might add a new issue to be able to select the text from that error window...

[image]

meee1, 2015-12-21:
thanks, fixed

closed 2015-12-21 by meee1

travmason, 2015-12-21:
Champion, will test in field today.

meee1, 2015-12-21:
you will need to download the beta mp to see that change

travmason, 2015-12-21:
I was going to have a crack at compiling it myself ;-)

travmason, 2015-12-21:
Harder than I thought it would be to find the beta downloads...where are
they hiding??

meee1, 2015-12-22:
goto help > update beta

travmason, 2015-12-22:
Duh. Thanks!