ArduPilot/MissionPlanner #1087: Extend interval between positions in Calibrate Accelerometers feature

heinduplessis, 2015-12-19:
When doing the accel calibration with large aircraft, it can take some time to properly align the airframe (TOP / LEFT / RIGHT / DOWN etc), by which time the command timed out on MP.

Can we extend the time taken for positioning the airframe?

rmackay9, 2015-12-21:
It should only progress to the next stage once the user has pushed the accept button so there must be something going wrong if it's progressing without you pushing any buttons.

heinduplessis, 2015-12-21:
Strange, I've done it multiple times this last 4 weeks, for myself and helping others and each time it's a race before it times out. Different computers and aircraft, both multi rotors and planes, APM and Pixhawk. I'll see if I can record a video.

meee1, 2015-12-21:
can you please provide more info about versions you are using?

meee1, 2016-01-10:
are you using usb or 3dr radio?

heinduplessis, 2016-01-10:
I've been away for a few weeks. I've been using the latest versions that were available in November and December 2015. I'm using the radio. I've spoken with a client yesterday which I'm advising, and he's also reporting that it's hardly possible to orient his large plane accurately before the timeout occurs. I'll do a  recording to illustrate. Thanks.

meee1, 2016-01-11:
does the issue happen via usb?
I believe this to a be a radio issue.

heinduplessis, 2016-01-11:
I haven't tried, I will do. Unfortunately most of the time the apm is deep inside the aircraft and the cable is too short - one can use an extension, but there'll be an additional pull on the usb slot on the apm / pixhawk.