ArduPilot/MissionPlanner #1088: Allow simultaneous RC and Joystick control

heinduplessis, 2015-12-19:
I see that if one uses a joystick on MP, RC control no longer works until the feature is disabled via the drop downs in the joystick setup screen in MP. This is rather dangerous since the telemetry can be lost and then one can no longer control the airplane at all.

Is it possible to allow simultaneous RC and Joystick control?

iskess, 2015-12-19:
I never used the joystick for this reason. If there were some way to instantly take over control with the RC I would feel much safer.

heinduplessis, 2015-12-19:
I must admin I had a slight panic attack reaching for the RC control and finding it not responding!

meee1, 2015-12-19:
this needs to go to the ardupilot list, as its a function of the autopilot not MP.

closed 2015-12-19 by meee1