ArduPilot/MissionPlanner #1090: Display servo outputs for channel 9 ~ 16

rmackay9, 2015-12-22:
Rob wanted to merge in this change to copter:
https://github.com/R-Lefebvre/ardupilot/commit/5cf703fbddd1c68593a70f1f357e09ff4cba2546

but during testing we found that mission planner seems to overwrite the ch1out, ch2out, etc with the values from channel 9~16.  Perhaps it's not checking the "port" parameter of the servo-output-raw message?

http://mavlink.org/messages/common#SERVO_OUTPUT_RAW

R-Lefebvre, 2015-12-22:
This is related to this Issue:

https://github.com/diydrones/ardupilot/issues/1968

meee1, 2015-12-22:
fixed

closed 2015-12-22 by meee1

R-Lefebvre, 2015-12-24:
Is this fixed in the current beta release?  I just tested it, and still not working though I see it is showing servo values for 1-16.