ArduPilot/MissionPlanner #1091: Feature request: Relay control

suraqis, 2015-12-24:
There's no relay controls in Mission Planer for switching relays on/off. It could be in "servo" control pane with "ON", "OFF" and "TOGGLE" buttons.

Jaaaky, 2019-06-14:
@meee1 Can this feature be implemented now?
Better to be like servo control tab. High(On)/Low(Off)/Toggle
mavproxy have this and it's very helpful.

meee1, 2019-06-14:
i would most likerly just append to the servo tab.

need to make sure i can even do it via mavlink first though

Jaaaky, 2019-06-14:
Nice. It's just as simple as MAV_CMD_DO_SET_RELAY long command. 
Here is how mavproxy do it;
https://github.com/ArduPilot/MAVProxy/blob/d7748c222ec0ed53b7b6b823f14e287f0e6c3df1/MAVProxy/modules/mavproxy_relay.py#L26

meee1, 2019-06-14:
cool, thanks for the ref.
should be easy then
i just needed to make sure its possible

Jaaaky, 2019-06-14:
I've already tested the mavproxy implementation and it's working fine.

meee1, 2019-06-26:
ive added this to the beta today