ArduPilot/MissionPlanner #1093: Better usage statistics

LorenzMeier, 2016-01-03:
@meee1 For the sake of Dronecode statistics it would be great to add / extend some of the statistics functionality you have.

It would be great to extend the URL with get params like:

```
http://s3bucket/binary?ap=<AUTOPILOT TYPE STRING>&id=<AUTOPILOTUID>&stack=<FIRMWARE STACK ID>
```

For MP statistics themselves, it would be great if it would open the MP website on boot to check for updates like:

```
http://updatesite.org/updatecheck/?id=<AUTOPILOTUID (= the first it has flashed)>&gcs=mp&version=<version>
```
and:
```
http://dronecode.org/updatecheck/?id=<AUTOPILOTUID (= the first it has flashed)>&gcs=mp&version=<version>
```

meee1, 2016-01-03:
Currently I already log these fields
MP Version
Board type (px4/px4v2/b2560/b2560v2)
Firmware Name (display name to user) eg APM:Copter 3.3 Quad
Culture (EN/FR etc)
Screen res
Screen colour depth
OS Version

LorenzMeier, 2016-01-03:
For the common URL we would want the GCS type and version and a way to identify the board (on upload) and the GCS installation (UUID of the system, if such a thing exists)

LorenzMeier, 2016-01-03:
Could you provide the definitive URL format you're proposing?

DonLakeFlyer, 2016-07-08:
@meee1 You log those fields on firmware download or on each connect to a vehicle?
@LorenzMeier I'm not sure about logging uuid of board without getting into privacy issues.

meee1, 2016-07-10:
On firmware download. As an internet connection may not be available on connect. I don't log any user identifiable info. And never will.