ArduPilot/MissionPlanner #1095: Host.InsertWP doesn't take altitude mode into account

kjeremy, 2016-01-05:
In my plugin:

I'm trying to set an unlimited loiter point into location 2 in the list
```csharp
PointLatLngAlt pt = new PointLatLngAlt(sensor.Latitude, sensor.Longitude, altitude);
Host.InsertWP(1, MAVLink.MAV_CMD.LOITER_UNLIM, 0, 0, 0, 0, pt.Lng, pt.Lat, pt.Alt);
```

but it doesn't take the altitude mode into account.  I think I want to do something like `FlightPlanner.setFromMap()` which takes the mode into account but am unsure as to the cleanest way to achieve what I want.

meee1, 2016-01-05:
when you say altitude mode, do you mean the function of verify height? or relative/absolute?

meee1, 2016-01-05:
was it just loiter unlimited you needed?

the plan would be to add
   else if (cmd == MAVLink.MAV_CMD.LOITER_UNLIM)
            {
                setfromMap(y, x, (int)z);
            }

to FillCommand :5029

kjeremy, 2016-01-05:
Yes just loiter unlimited.  Thanks I'll try that.  If it works I'll submit a PR.

meee1, 2016-01-05:
I already have the change pending, so all good, I will commit.

closed 2016-01-10 by meee1