ArduPilot/MissionPlanner #1099: STorM32 Gimbal Controller: Mavlink Parameters not displayed

olliw42, 2016-01-10:
Hey

I've recently updated to MP 1.3.34 (I can't tell for sure, but I may have come from 1.3.2x, it was a version I used 1/2 year ago)

when attached to a STorM32 gimbal controller, configured for SYID = 71 and COMPID = 67, one can see that MP reads in the MavLink parameters after hitting CONNECT, and the HUD correctly shows the camera's attitude, but the screen CONFIG/TUNNING nearly always remains blank
sometimes the CONFIG/TUNNING screen shows up as expected, and the parameter tree is available and can be browsed, but the parameter values are nonsense (I could not determine the circumstances, but disconnecting and reconnecting seems to make that)

This is so also when I flash an 1/2 year old firmware (v0.86e) on the STorM32 controller, which was known to work previously.

THANKS
Olli

meee1, 2016-01-11:
when you transmit paramaters to the gcs what format are they in? and what param format type are you sending?
ardupilot incorrectly uses the float type, so all param a floats, even though the type is set otherwise.
mp had a change recently to follow the spec if it was connect to a non ardupilot system. so if the param type is sent to something non float, then the data type should follow the type, not float.

olliw42, 2016-01-11:
thx a lot for looking into that
the parameters are either UINT16 or INT16, and that's also the Mavlink formats which are send (3 or 4)
I'm not sure how a parameter format issue would make the screen to remain blank in nearly all cases
I should say, I'm not using the original MAVlink package but did my own thing, I didn't took AP as guide at all, in my implementation I simply followed the descriptions on the MAVlink webpages with occasional peeks into to original MAVlink code, so I would assume I'm compatible with that (if nothing has changed there in the last few months)

olliw42, 2016-01-16:
hey
so, I spend some more time on the issue and rolled back firmwares

1.3.28: 
works perfectly
no blank screen after connect 
parameter values are OK (it doesn't get negative int's correct, but this had been "ever" like that)
1.3.29, 1.3.30:
with 1.3.29 the blank screen issue seems to have been introduced
it seems to be like this: whenever I do a connect and have NOT visited the CONFIG/TUNNING tab before, I'll get a blank CONFIG/TUNING tab after having connected
so, looks pretty much like some init variable thingy
1.3.31, 1.3.32, 1.3.33, 1.3.34:
blank screen issue continue to exist
with 1.3.31 the incorrect parameter value issue seems to have been introduced
in the release notes I didn't found much indication of what the changes really were
from your comment it's just clear that something changed

I could speculate: In C one would not set a mavlink parameter value field like floatvalue = intvalue; but like memcpy(floatvalue,intvalue,size);. But if so this seems not happen in e.g. mavlink_msg_param_value_pack(), as I can't see that something has changed here (it's still packet.param_value = param_value;, https://github.com/mavlink/c_library/blob/master/common/mavlink_msg_param_value.h#L60)
???

meee1, 2016-01-18:
normally param_value is a union.

the easiest workaround, if you ijmplemented using the old structure is to just set param_type to float. and then it will work as exepected.

closed 2016-03-08 by meee1