ArduPilot/MissionPlanner #1111: Waypoint Variables.

MusabAlaqeede, 2016-01-23:
I am trying to update the waypoints on the flight plan through a webapplication I have designed, I have managed to connect mission planner to the database on the webapplication and I am able to retrieve the data (Longitude and latitude). however I have failed to find the variables that make the waypoints. I might have been looking in the wrong files to begin with... please help direct me. 

Attached below is an image  how I want the data to appear after updating the database.  
[image]

meee1, 2016-01-24:
your question is way too vague.

MusabAlaqeede, 2016-01-24:
Which part of the question are you unclear on ?

meee1, 2016-01-28:
look at the function 
public void AddWPToMap(double lat, double lng, int alt)
or
public void setfromMap(double lat, double lng, int alt, double p1 = 0)

closed 2016-01-28 by meee1

MusabAlaqeede, 2016-02-03:
Thank you so much for your kind response, I am trying to set the 'lat','lng' & 'alt' from a different program as an API, how would I approach this matter as I am new to C#.