ArduPilot/MissionPlanner #1112: Radio calibration screen and AUX functions for CH9 to CH12

lvale, 2016-01-25:
Would be helpful to also have at least 12 channels displayed on the Radio Calibration screen. People don't know they can use channels greater than 8, because of this.

pristinesource, 2016-03-25:
I think this is a limitation of MAVLink.  Everywhere I have read in the MAVLink docs it only lists up to 8 channels.  If I found out different, I will try and get it updated for you and send in a Pull Request to @meee1.  I also could benefit from being able to utilize more channels.  I think a lot of consumer UAV systems go up to 16 total channels.

pristinesource, 2016-03-25:
Looking at the MAVLink documentation, it appears it can read up to 18 channels <https://pixhawk.ethz.ch/mavlink/#RC_CHANNELS>, but can only override/write the values for the first 8 <https://pixhawk.ethz.ch/mavlink/#RC_CHANNELS_OVERRIDE>.  I am not sure why this is a limitation of MAVLink, but I am certain this will require an update in MAVLink before MissionPlanner can support it.

pristinesource, 2016-03-25:
Yep, just as I had suspected.  It looks like the MAVLink firmware on the UAVs actually have the limit of only setting the first 8 channels.  Here is the code for the PX4 firmware where it sets those channels and you can see that we only have access to channels 1-8.  <https://github.com/PX4/Firmware/blob/6d42595c795adc1fa5729a49c24f25f75634d1f6/src/modules/mavlink/mavlink_receiver.cpp#L1114>

pristinesource, 2016-03-25:
Great news! I found a way to support the additional channels using the MAVLink MANUAL_CONTROL command <https://pixhawk.ethz.ch/mavlink/#MANUAL_CONTROL>.  It looks like this command is specifically designed for taking input from joysticks and it supports additional channels via the "buttons" field.  The only problem is that it only supports a up/down state for channels 5-16+ and cannot do an analog input on those.  This means that channels 5-16 would only be up/down button states.  I think I can do it as a hybrid of what is currently there to support analog input in channels 5-8 and using the MANUAL_CONTROL command to do the input for channels 1-4 (flight controls) and the button inputs for 9-16. I will get started on this right away ;).

I am really blowing up this thread, lol!