ArduPilot/MissionPlanner #1122: quick key for switching between quick,actions... tabs

haogodent2, 2016-02-02:
This is a feature request. It would be nice if the tabs are slaved to different number values. This way we can quickly switch between tabs such as quick and actions which I find myself doing very often.

meee1, 2016-02-02:
this I think could happen.
what key shortcuts would you suggest?

haogodent2, 2016-02-02:
I think using the 1,2,3,4,5... number keys would make sense here since there are quite a few sequential tabs and they are not currently mapped to anything.

meee1, 2016-02-02:
what about the modifier key? ie shift/ctl/ etc.

haogodent2, 2016-02-02:
I see that that ctrl+tab already scrolls through the different tabs. Do you mean ctrl+1, ctrl+2? That would work for me.

meee1, 2016-02-02:
that could work,

currently f2-f4 switch the main views
and f12 connects.
so ctrl-1... sounds like a good idea

haogodent2, 2016-02-02:
Ah, didn't know the F1,F2 shortcut, that's actually pretty cool for switching frequently between flight, plan, and config.

iskess, 2016-02-02:
Is there a list of shortcuts published anywhere? It would be great to have it somewhere in MP for reference.

meee1, 2016-02-16:
Main Screen
F2 - FlightData
F3 - FlightPlanner
F4 - Tuning
F5 - Refresh full param list
Control-F - Temp screen
Control-X - sysid selector
Control-T - Blind connect

FlightData
1 - tab 1
2 - tab 2
3 - tab 3
4 - tab 4
5 - tab 5
6 - tab 6
7 - tab 7
8 - tab 8
9 - tab 9
0 - tab 10

FlightPlanner
Control-Z - undo
Control-O - open wp file
Control-S - save wp file

AutoWP Grid
Control-O - open grid file
Control-S - save grid file

closed 2016-02-16 by meee1

iskess, 2016-02-16:
Thanks Michael. Could this be listed under the help tab?