ArduPilot/MissionPlanner #1123: ability to remove/move tabs

haogodent2, 2016-02-02:
this is a feature request. It would be nice if there was a list of check boxes for the different tabs such as quick, actions ...  to turn them on/off OR be able to drag the tabs around to prioritize. It's inconvenient to switch from quick or action tab to messages tab by scrolling using the left and right button.

Thanks,
Hao

magicrub, 2016-02-02:
I checked into this and .Net does not have any API to set visible for ControlTabs. The only way to do it was to remove/add the tabs. So, on start we'd have to cache them somehow and so removing is easy and adding a tab would be adding the cached index. However, when I tried this there was a ton of code which assumes the tabs are always there so we crash in a zillion different areas that wrote to those tabs. They would have to change to either not write or write to the cached version.

meee1, 2016-02-02:
if I change the default, whats is your preferred order?

haogodent2, 2016-02-02:
personally I use the mission planner mostly for debugging/test flying purposes so I use the messages and DataFlash Logs tabs quite a bit. I don't use the Gauges tab much since its a bit redundant with the HUD already in place. I would prefer to have the messages moved up and perhaps gauges moved back a little.

meee1, 2016-02-02:
I might leave the order at the moment, in favour or the control-1-9 idea. this way there will be predictable.

haogodent2, 2016-02-02:
sounds good. thanks.

closed 2016-02-02 by haogodent2