ArduPilot/MissionPlanner #1126: Laser altimeter geotaging csv

raptortrias, 2016-02-05:
Is it possible to add functionality to create a csv file with sonnarrange [used with laser altimeters] while geotagging images in mission planner. This way the above ground data can be imported in software like agisoft to increase model accuracy.

 Thanks

meee1, 2016-02-08:
do you have a log of sample data I could use to test?

raptortrias, 2016-02-11:
This log file deosn't have any trig cam messages, but the lidar was activated. The lidar altitude is at CTUN Salt  . Ideally during georeferencing we could get the sonar alitude in a csv for every image together with GPS coordinate
[10_02_log.zip](https://github.com/diydrones/MissionPlanner/files/126623/10_02_log.zip)

raptortrias, 2016-03-20:
Michael,

 If we replace RelAlt with Salt in the geotagging code, would this work for exporting the GPS altitude as well as the above ground altitude during geotagging?

 Thanks

raptortrias, 2016-04-23:
Michael,

 I don't have any other log files as the system is with the customer now, but if it would have been possible to modify the script so that it can pull the Salt (sonar altitude) during geotagging it would be great.

the location.txt would be the best place to have this.

 Thanks

nicksargeant, 2016-06-17:
Hi Michael, 
I too very much need this functionality but it appears to only work for Copter logs because rangefinder distance is stored as CTUN.SAlt on Copter and RFND.Dist1 on Plane. Any chance on a patch?
Kind regards,
Nick

meee1, 2016-06-17:
Added @nicksargeant

closed 2016-06-17 by meee1

raptortrias, 2017-04-04:
@meee1  thanks very much