ArduPilot/MissionPlanner #1127: Mission planner 1.3.35 Auto Analysis

JeyPi, 2016-02-05:
When I try to auto analyse a log (arduplane 3.5.0 Pixhawk) I get a bad input message.
Win. XP7
[Log](https://www.dropbox.com/s/nqmjqglkhesjsmq/1.BIN?dl=0)

stik79, 2016-02-05:
Same thing Win10. Attached log.
NpyAccessLib.dll exists in DLL directory within MissionPlanner.

[NpyError.txt](https://github.com/diydrones/MissionPlanner/files/119986/NpyError.txt)

stik79, 2016-02-07:
As a quick workaround launching missionplanner from within the DLLs directory (i.e. ..\MissionPlanner.exe) picks numpy up and works - which is strange because if I do a print sys.path it has added the DLLs directory running normally (i.e. from the application root) too...

Unfortunately I'm not enough of a python/.Net guru to diagnose much more :(

RestlessC0bra, 2016-02-10:
Same thing here. Win8.1 Gives me error messages

The type initializer for 'NumpyDotNet.NpyCoreApi' threw an exception

and then

Bad input file

I can't analyze any logs!

Hankster1228, 2016-02-11:
Same thing here. 

 Windows 7 gives me the same error message when I try to auto analyze a log.  I have the most recent version of Mission Planner 1.3.35 build 1.1.5878.12941 downloaded 3 pm central February 11, 2016.  Error message is: The type initializer for 'NumpyDotNet.NpyCoreApi' threw an exception followed by a small dialog box stating "Bad Input File".  The file can be downloaded and analyzed but just not auto analyzed.

meee1, 2016-02-13:
as a temp fix, please delete NumpyDotNet.dll from the MP folder.

it should remain in the "dlls" folder however.

Hankster1228, 2016-02-13:
Thanks! That fix worked.

JeyPi, 2016-02-14:
Well not for me. After that I get first -  **System Exit** message and than **Bad input file**

Hankster1228, 2016-02-14:
I went to my Mission Planner program folder (*c:\program files
(x86)\Mission Planner\*) and renamed the NumpyDotNet.dll to NumpyDotNet.DAL
and it fixed the problem.  I renamed the DLL rather that erasing it just in
case I screwed up and deleted the wrong dll.  I'm running Windows 7.

Good luck!

meee1, 2016-02-16:
the current Beta MP should resolve this issue
https://github.com/diydrones/MissionPlanner/commit/fe624427242cda810c5fc64030558c5cae70a3c0

closed 2016-02-16 by meee1