ArduPilot/MissionPlanner #1129: Really need improved cam trigger (cam_trigg_dist)

Gun-jam, 2016-02-08:
Hello
we photograph many acres per day (1000+) attempting to create a clean layout for large fields is a nightmare from a camera trigger perspective. 

for smaller fields I use digicam control but I can only collect about 300 images using this method as each camera trigger is based on a waypoint and each waypoint uses 2 waypoints worth of data one for the location and one for the action (fire the camera) this burns up the 700 waypoints available in the pixhawk immediately. 

Attempting to use Cam_trigg_dist solves the waypoint problem but creates a huge pile of junk photos problem leaving me at the end of the day with a 1000+ worth of images that are of no use but take up processing time (wasted work). This is because once the UAV hits wp 1 it just fires the camera every say 35m it does not pay attention to the polygon boundary. 

It appears that of recent an attempt was made to solve this, an option called "break up starts"  this does not work much better because again it ignores polygon. I use 170m of leadin and overshoot to allow the uav to get back on course it continues to take photos during the leadin and overshoot segments in this location the UAV is 1: outside area of interest and 2: at some point in a turn. This results again in huge amounts data in the form of useless pictures.  

I have been solving this by adding waypoints and camera start values to the border of the polygon this takes mission creation time from 5 min to 45+ for large fields this also requires adding altitudes...it is not fun and easy to screw up.

The solution is to have "breakup starts" option work in the following way:
During mission creation with cam trigg dist selected and "break up starts" selected Mission Planner would apply a waypoint  when the flightpath crossed the polygon border. If the UAV is flying into the border then cam_trigg_dist has a positive value (say 35m) and starts the camera. If the UAV is flying out of the polygon then set cam_trigg_dist to a 0 value and shut off the camera.

The one issue I can think of would be if a flightpath were directly over a polygon line and they ran parallel...The solution would be to make that "non allowed" either mission planner would shift that one leg a few pixels one way or the other And/or warn the user that in this case that cant occur and adjust photo sidelap to change lane width so the flightpath and the polygon line are not on top of eachother.

If this could be implemented it would save many hours of wasted time....I doubt im the only one who is having to deal with this.

Thank you for looking into this and for all the work you have provided the community.. It is appreciated.

meee1, 2016-02-08:
ive resolve the leadin issue with a patch. I will upload a beta MP in the next 10 mins

meee1, 2016-02-08:
the issue with the line, over a polygon boundry is a lot of people change the angle to suit wind conditions. I dont feel it would best archive the outcome you are asking.

Gun-jam, 2016-02-09:
Im not a programmer by any means nor do I know the limitations of what you have to work with but I try my best to see things from the others perspective...really all that Is required is that waypoints and camera triggers appear at the polygon boundary without me having to add them so photos are only collected within the polygon using distance methods. This would solve a lot of my problems!

Thank you again for looking into this!

meee1, 2016-02-09:
yep, that's the change I made this morning. and if you download the beta MP, it should be like you mention.

help > update beta.

Gun-jam, 2016-02-09:
I was able update to the beta that you provided 1.3.35 build 1.1.5883.11849

However i dont see any changes to the way things work...Im not sure if im missing a step or there is some other issue. 

This how they look when I try them:
[image]

This is how they look when I set them up manually and what I expected "break up starts" to do after the beta that you just posted.
[image]

Let me know what you think or what im missing.

Thanks again!

meee1, 2016-02-13:
please try the beta again, what your "expecting" is what you should be getting in the beta

Gun-jam, 2016-02-14:
Yes I had to remove mission planner, reinstall then download the beta and it does indeed work exactly as I need it to...This will save us many hours and improve our efficiency in some case by double and reduce all the junk work of downloading and sorting out junk photos. Its nice to know I can take more than 300 photos now if I need to.

Thank you for your help it is greatly appreciated.

closed 2016-02-16 by meee1

IronDome2, 2016-03-18:
Waaw Gin-Jam

i thought I was the one and only having this frustration. 
Really, if this is resolved I would be a happy man.

Thanks for sharing

iskess, 2016-03-18:
It was resolved. That's why the issue is closed. Do you have the latest revision?

magicrub, 2016-03-18:
The next release, in the next week or so, has an improved can trigger.
There is an additional improvement we have in mind for a later release to
make it micro second accurate.

iskess, 2016-03-18:
What improvement to the trigg is coming in the next release?

magicrub, 2016-03-19:
Reducing the latency of logging and io pin output. It went from many ms to
a couple ms and next update will be in the microsecond range.

These are firmware changes though, not mission planner.

IronDome2, 2016-03-19:
Sorry for re-opening a closed thread.

I do not have the latest release yet, I have asked my manufacturer to test it before i can use it for Warrant purposes.

What about footprint showing covered area, this would be helpful when planning mission over a large area.

Thanks in advance

magicrub, 2016-03-19:
Normally I would say you should open a new issue for a new topic but there
is already one for that.

onurboy, 2016-04-04:
Very important and helpful feature. Thank you for this enhancement.