ArduPilot/MissionPlanner #1132: Custom firmware .px4 file reports no end flag when uploading

ycai47, 2016-02-11:
I was trying to add IR Lock precision landing on my quadcopter and I followed [http://dev.ardupilot.com/wiki/building-px4-with-make/](url) to make the .px4 file after I modified the APM_Config.h file to enable precise landing and build from master branch of ardupilot as suggested by this page, [https://irlock.readme.io/docs/download-links](url). However, it keeps giving me an error like this:

> The hex file does no contain an end flag

And then followed by:

> Failed to read firmware.hex: No end flag in file

Is there anything that needs modified in Utilities/Firmware.cs or it is rooted somewhere else?

meee1, 2016-02-11:
what other usb devices do you have plugged in?
I would try unplugging other atmega devices and try again.

ycai47, 2016-02-12:
I tried unplugging everything, even restarting. I finally uploaded with APM Planner 2.0, so that's probably special about Mission Planner.

meee1, 2016-02-13:
can you post your missionplanner.log file for when the error occurs. as for some reason something connected to your pc is being identified with a vid and pid of a apm2 style device.