ArduPilot/MissionPlanner #1134: Feature request - sum of segment lengths

charleslinquist, 2016-02-12:
I fly mostly autonomously.  And although Mission Planner gives me the length of the segments between each waypoint, it doesn't sum them.  It would be nice if the total length of all segments was shown at the bottom of the list of waypoints.  That way, I would be less likely to send the craft further than it could safely go.

Better yet would be a warning when the total length of the segments exceeded a user-set value, and even better would be a rough approximation of how much battery life would be consumed by the time it reached each waypoint.  Climbing and descending battery usage could be estimated to come up with a more accurate figure.

meee1, 2016-02-13:
in flightplanner in the top left there is a total of the planned mission, is that not what you are looking for?

also in flight data the bar at the top of the map with the yellow dots?

charleslinquist, 2016-03-08:
I apologize!  I didn't see that there.

But I do have what I think is a legitimate request:

Filter the current reading through an averager or smoother.  The current 
readings bounce around so much they are difficult to interpret.  You 
could give the user a choice as to the degree of smoothing, so they 
could still see narrow spikes if they want to.

Also, the DROOP parameter is not well documented.  I don't know what it 
is supposed to be, but if it is supposed to be an indicator of the  
battery ESR, I don't know how it is used.  And if it really isn't used, 
it could be changed into a "corrected" battery voltage.

Most LiPos have an ESR of around 6 milliohms per cell.  Using that 
figure, a four cell battery loses 24mV per Amp in that ESR, so the 
actual cell voltage is higher than the terminal voltage by that much. 
The curves I find on line that show discharge voltage vs. current would 
indicate that the lower voltage at higher current draw is nearly 100% 
explained by simply considering the ESR of the battery.

At 30A, the voltage reading of a 4 cell battery is 0.72V low (terminal 
vs cell), so the voltage could be adjusted upward by the smoothed 
current, thereby giving a more accurate indicator of battery 
condition.   My readings show that a figure of 6 milliohms per cell is 
close to what  the average new battery has.  As the battery ages, the 
ESR goes up and the error increases.  Defaulting to 6 milliohms/cell is 
a good starting point, and if the user wanted to change that, they could.

I realize that the right place for this adjustment is in the controller, 
but nonetheless, providing this "corrected" reading as well as the value 
provided directly by the controller is just one more data point. To me, 
battery condition is probably the most important factor in flying. The 
program could even dynamically calculate battery ESR by looking at 
voltage at two closely spaced intervals while at the same time watching 
the current. If the current changed by 10A or more within a few seconds, 
the voltage before and after the change could be noted and the ESR 
calculated.

Charles Linquist

Check out my multirotor blog at http://calmultirotor.com