ArduPilot/MissionPlanner #1135: DO_Digicam_control does not work with 3.5 (no way to take pictures)

Gun-jam, 2016-02-13:
Hello

With pixhawk running 3.5 firmware the DO-DIGICAM_CONTROL does not function. This is my primary method of image collection.

The issue appears to be with Shutter CDM in the DO_DIGICAM_CONTROL options it is by default 0
Apparently now it must be a non zero value.

Setting this value to 1 causes the UAV to capture a photo it also causes the waypoint line to head way off the map im assuming to GPS 0,0 once the flight plan is uploaded...before upload it looks okay in the flight plan window....in the flight data window its no good.

The UAV still executes the flight plan as it should but now there is no reference because Mission planner  show flight path like off cost of Africa or something. 

I usually shoot 280+ images per flight...I can not set Shutter CDM value to 1 for each trigger location
Having MP show flight paths heading to Africa is a bit of a danger because now I cant tell if its a bug or the UAV is really going to Africa. 

Im hoping you can come up with some type of solution for this.
Thanks
-Nick
[image]
[image]

Gun-jam, 2016-02-13:
further note... this occurs when using "Auto WP" to draw "survey (grid)" inside a polygon. and using DO digicam control as the method of trigger 

if you just point to a few places and add "do digicam control" using add below...it works fine.

meee1, 2016-02-13:
thanks for the bug report, and the details you provided.

Gun-jam, 2016-02-13:
any idea when this would be available to try out?

Thanks

meee1, 2016-02-13:
now, should work.

Gun-jam, 2016-02-14:
Okay so first of all looking back at issue 1129 (the distance trigger method) this does indeed work exactly as I need it to. The problem is I had to uninstall mission planner then reinstall then download the beta.

I did that exact same step now and everything seemed to install just fine but when I create a mission using survey (grid) and look at "shutter cmd" the value is still zero... I did not fly to test if it actually works yet but my feeling is NO...As far as I can tell if shutter cmd is 0 it wont take a picture.

Should I expect to see a 1 value for shutter cmd or did you solve the issue using a different method?

Thank you for your help!

meee1, 2016-02-16:
did you update to the beta MP?
this change is only in the beta MP.

closed 2016-03-08 by meee1