ArduPilot/MissionPlanner #1136: Mission planner 1.3.35 - Flight data map window does not show route sometimes

JeyPi, 2016-02-13:
I might be doing something wrong, but the flight data screen does not show waypoints in its map window as soon as the waypoint distance exceed some measure.
For example if my planned route is within the radius of 2km it is fine. But if I plan a route that is in a radius of 10 Km, the waypoints will not show up. If I want to see the plane following the route I must go to the flight plan window

meee1, 2016-02-13:
do you have an example?
also do you have "auto pan" turned on?

JeyPi, 2016-02-14:
Well I can post photographs, or the waypoint file. The mentioned behavior occurred
 several times. It should not be a problem to replicate it.
Here is a missing waypoint. The picture shows the route in the Planning window and below in the flight window
[image]
The second picture shows no waypoints at all in the flight data window while they are planned in the flight plan window
[image]
These will no show up in the flight data window.
[waypoint example](https://www.dropbox.com/s/34k78uyuy1jxsbu/APOK.waypoints?dl=0)

meee1, 2016-02-15:
normally flight data will only show what has been received or sent to the autopilot. ie if you load a mission and never send it it will never show up.

JeyPi, 2016-02-16:
Confirmed and closing

closed 2016-02-16 by JeyPi