ArduPilot/MissionPlanner #1137: While clicking on table in firmware log viewer, get 'System.StackOverflowException' occurred in System.Drawing.dll

rrr6399, 2016-02-14:
Over the last few months, I've been getting  'System.StackOverflowException' occurred in System.Drawing.dll when I am using the firmware log viewer. Many times when I click on a row in the data table, the GUI locks up and eventually it crashes with the overflow error. I am also getting this error during startup, but not sure if it is related:

Message: Managed Debugging Assistant 'LoaderLock' has detected a problem in 'C:\MissionPlanner\bin\Debug\MissionPlanner.vshost.exe'.
Additional information: DLL 'C:\Windows\assembly\GAC\Microsoft.DirectX.DirectInput\1.0.2902.0__31bf3856ad364e35\Microsoft.DirectX.DirectInput.dll' is attempting managed execution inside OS Loader lock. Do not attempt to run managed code inside a DllMain or image initialization function since doing so can cause the application to hang.

Here is a snippet of the stack trace that happens during the stack overflow;

System.StackOverflowException was unhandled
Message: An unhandled exception of type 'System.StackOverflowException' occurred in System.Drawing.dll
System.StackOverflowException was unhandled
Message: An unhandled exception of type 'System.StackOverflowException' occurred in System.Drawing.dll
     [Managed to Native Transition]   
>    System.Drawing.dll!System.Drawing.Font.GetHeight(System.Drawing.Graphics graphics)    Unknown
     System.Drawing.dll!System.Drawing.Font.GetHeight()    Unknown
     System.Drawing.dll!System.Drawing.Font.Height.get()    Unknown
     System.Windows.Forms.dll!System.Windows.Forms.DataGridViewRow.DataGridViewRow()    Unknown
     System.Windows.Forms.dll!System.Windows.Forms.DataGridViewRow.Clone()    Unknown
     System.Windows.Forms.dll!System.Windows.Forms.DataGridViewRowCollection.this[int].get(int index)    Unknown
     System.Windows.Forms.dll!System.Windows.Forms.DataGridView.DataGridViewAccessibleObject.GetChild(int index)    Unknown
     System.Windows.Forms.dll!System.Windows.Forms.DataGridViewRow.DataGridViewRowAccessibleObject.Bounds.get()    Unknown
     System.Windows.Forms.dll!System.Windows.Forms.DataGridViewRow.DataGridViewRowAccessibleObject.Bounds.get()    Unknown
     System.Windows.Forms.dll!System.Windows.Forms.DataGridViewRow.DataGridViewRowAccessibleObject.Bounds.get()    Unknown
     System.Windows.Forms.dll!System.Windows.Forms.DataGridViewRow.DataGridViewRowAccessibleObject.Bounds.get()    Unknown
....
....5000 calls later, get this:
The maximum number of stack frames supported by Visual Studio has been exceeded.

meee1, 2016-02-15:
do you do regular windows updates?

rrr6399, 2016-02-18:
Yes. But, I will verify that the updates are recent.

rrr6399, 2016-02-18:
Looks like I had some downloaded Windows updates that weren't installed. MP works much better now. I can zoom reliably and it hasn't locked up so far despite how many times I click around in the Log Browser table even with a 100 MB log file. I'll go ahead and close the issue.

closed 2016-02-18 by rrr6399