ArduPilot/MissionPlanner #1138: Problem with sitl on mission planner 1.3.35

Fortesque0, 2016-02-14:
Hello:
I cant set home position on sitl. I used on windows 7, windows 10 and I have the same issue. The sitl for anyone vehicle it's always lng -214,74.... The latitude changes, but the longitude didn't. If I arm the vehicle (copter, plane, heli....) the simulated model starts to move but it can't change the long. It goes up or down through the map.
Any idea?
I use administrator to run mission planner, I change source maps, but always the same issue.
I downgrade mission planner and the same problem.
Thanks in advance

Fortesque0, 2016-02-16:
Hello:
I forget to comment that I am from Spain, and this is the problem. Finally, I found the answer. The region configuration on windows it's the solution. I change the windows configuration from Spain to English US.... Eureka! Now SITL works. So it's possible to fix this on next Mission planner versions?

magicrub, 2016-02-16:
There was a PR ( https://github.com/diydrones/ardupilot/issues/3586 )that fixed a problem yesterday. Was your problem regarding compiling or something else?

meee1, 2016-02-16:
thanks Fortesque0, ive merged a fix for the language issue

closed 2016-02-16 by meee1

Fortesque0, 2016-02-18:
Hello Michael,
At first, thanks for your work. It's really wonderful. 
The problem is not the language. Concretely it's the data type form
[image]
Changing this from Spain to English us sitl works.
Other question (I am newer), there is graphical integrated simulator in mission planner (old version has flightgear incorporated) or I have to download one (jsbsim,flightgear,xpjane....), and how I can configure it ( I been searching on ardu protect wiki but there is not information running from mission planner under windows). I found something for xpkane  from Google, but for old versions of mission planner.
Thanks in advance