ArduPilot/MissionPlanner #1139: BSOD on Firmware update attempt.

Charlweed, 2016-02-15:
Has happened 3 times now, upon clicking the "OK" button when attempting to upload APM:Copter V 3.3.2 Y6 firmware. Problem is new to 1.3.35 (build 1.1.5878.12941)  mission planner, problem never happend on previous release of Mission Planner.
Target  'copter is a 3DR Pixhawk.
Host is a high end Windows x64 system (for 2016 anyway) , I expect the only hardware note is that the USB connection is 3.0 :
Intel Core i7-33960x @3.30GhZ
16.GB Ram
Motherboard: chipset x79; ASUS RAMPAGE IV Black edition
OS: Windows 7 Professional.

I would prefer not to upload a dump file, but I can, if it will actually help. Here is the Win 7 description after reboot.

Problem signature:
  Problem Event Name:	BlueScreen
  OS Version:	6.1.7601.2.1.0.256.48
  Locale ID:	1033

Additional information about the problem:
  BCCode:	44
  BCP1:	FFFFFA801D4D4810
  BCP2:	0000000000000EC0
  BCP3:	0000000000000000
  BCP4:	0000000000000000
  OS Version:	6_1_7601
  Service Pack:	1_0
  Product:	256_1

Files that help describe the problem:
  C:\Windows\Minidump\021416-15147-01.dmp
  R:\chymes\TMP\WER-1158260-0.sysdata.xml

meee1, 2016-02-15:
MP doesn't have the capacity to cause a blue screen. however the device drivers/ hardware does. I would start looking down those paths.

meee1, 2016-02-15:
BCCode: 44   0x00000044

 This is almost always a driver or something causing a driver failure or even an interaction 
 of two drivers. Have you added new hardware lately or updated drivers - also check in 
 Control Panel - Windows Updates to see if any drivers have been updated there.

closed 2016-02-15 by meee1

Charlweed, 2016-02-15:
OK, will recheck my drivers and won't reopen. For the record:
Attempted to replicate bug on low-end laptop, Celeron N3050, 2GB ram. No BSOD, because upload cannot be started. Failure to connect error using powered USB 3.0 Hub. Error is “The given port name does not start with COM/com or does not resolve to a valid serial port.”
I cannot connect directly to built-in USB 3.0, as target Pixhawk. power requirements may be too high.

magicrub, 2016-02-15:
USB 2.0 supplies 500mA and that will per the pixhawk and a few peripherals
just fine. USB 3.0 supplies 900mA by default.