ArduPilot/MissionPlanner #1147: New Flight Mode

bitusbates, 2016-02-18:
Hello,
I'm trying to do a new flight mode, in order to facilitate passage of ambulances and other emergency vehicles. That's why I want to create a flight mode that do that my drone stays always a certain distance from me or a vehicle (for example 100 meters). I will use a telemetry module with my mobile, and a U-Blox  6M NEO GPS in my drone. The idea is to set some waypoints with the Tower App, and then through this "New Flight Mode", do that the drone stays always these 100 meters (radius) from me or vehicle. See please the image below to undestand the situation easier 
[image]
Could someone help me to create this new mode?
Thank you so much,
David

meee1, 2016-02-21:
you cant do what you want already with follow me?

JimiJoplin, 2018-05-11:
hi @meee1 can you post some links to the follow me mode? I have to implement it on a rover but i can't find references.
thanks