ArduPilot/MissionPlanner #1148: Ubuntu/linux/Mac* packages for MissionPlanner

ustramooner, 2016-02-21:
I've been playing around with MissionPlanner to get it working for Linux. I've had some luck (not 100% yet) - but I'd like to share my work. It's not ready for merging yet, but would like to hear some feedback on it.

https://github.com/ustramooner/MissionPlanner/tree/MONO-linux

The kind of stuff I've worked on is:
* getting xbuild to work (mostly just file naming - case sensitivity)
* got a basic debian packaging system going (which will help with support - i see a lot of issues are due to people using older mono versions or not installing certain packages)
* sqllite - i'm a bit confused how the Mono.Data.SqliteClient package works, as I understand this is old
* I've had to delete out the joystick stuff - but the plan would be to abstract it with some yet-to-be-discovered library
* moving out the config and files that are written to to a configuration directory. linux install paths are read-only (and i think this is good practise to not write to your install dir). Using MainV2.UserPath.

bigger issues i've come across:
* the joystick thing
* NumpyDotNet - what's the plan here - it doesn't seem to be supported anymore, and is only available on windows. I tried porting it, but got stuck on something with the numerictypes.py
* I haven't dug any deaper into the hidden functionality and layout issues, but it's usable for me now.
* random lock-ups - this makes it a bit dangerous to fly with so I always have APM Planner ready to go. For example I come across an issue where the screen locks up - i haven't figured it out exactly but seems to be related to a multi-threaded timing issue with X windows - when putting sleeps in certain places it seems to be better/fixed - several places still have this isssue.

currently left to do:
* sort out joystick code
* ideally do a mac package (i develop on linux - but normally take a mac to the field when flying). most of the fixes here help with the mac versions)
* be nice to filter the serial devices a bit better (currently it shows all the /dev/tty* devices - where as APM Planner manages to filter down to only more relevant devices).

Looking forward to hearing some feedback!

meee1, 2016-02-22:
you seem to have got the major issues identified, i do know of some others
mainly datagridview, and its lack of support for virtualmode.

meee1, 2016-02-27:
ive also tested running windows built mission planner under ubuntu and mint. it does run, and is usable.

meee1, 2016-02-28:
ive merged a few of your commits. the ones that made initial sence. There are a few commits that are corrections to other commits, which i skipped.

ustramooner, 2016-02-28:
@meee1 thanks. 

Main problem is still the Joystick - I've ordered one so i can work on it myself. once that part is done then i think we can run through the entire patch to get it merged.

Nitrosito, 2016-02-29:
nice job !

We must have a  GNU/Linux version of this software!

meee1, 2016-02-29:
if you install mono, and use the zip version of mission planner you can already use it.

minus a few features

kikislater, 2016-03-01:
Last time I tried, Calibration screen does not works, because I think due to use of directx for this part of the software. Should it be solve ?

kikislater, 2016-03-01:
Last time I tested it, Calibration screen is not working maybe due to use of directx. Should it be solve ?

ustramooner, 2016-03-01:
this patch is more about building on linux and packaging. there are a few minor issues fixed, but MP did already work

Nitrosito, 2016-03-02:
This ocurred when open MisionPlaner10.exe with mono
[image]

meee1, 2016-03-02:
why do you have such an old version?

Nitrosito, 2016-03-02:
ups!
I donwload from google search "APM zip install"
Where are the zip files?

kikislater, 2016-03-02:
It's here : http://ardupilot.com/downloads/?did=83

Raviu8, 2018-07-25:
Hi,
There are updates and I think this could be done.
I'm not a developer but a Linux user. I've tried to get things working as much as possible to the least number of errors.
Here are the steps - hoping it could help someone to get a native Linux version.

At first to get almost ready environment the best is to use the method here
https://www.mono-project.com/download/stable/#download-docker
Using any Linux system with docker. The docker image is based on debian 8 - good.

After running `docker run -it mono:latest bash` and getting it working for the 1st time you can Ctrl+D then run `docker ps -a` and get its name - the right most column - .
Then you can run `docker start -i Name-name` to start the same docker image again.

You'll find that you've latest mono with the new msbuild package - xbuild is deprecated.

Run the following command to install needed packages;
`apt install libunwind8 openssl git wget`

Then go here https://www.microsoft.com/net/learn/get-started-with-dotnet-tutorial and install .net-sdk, choose debian 8 and follow the steps.

Then git clone MissionPlanner repo.

Then to build I've tried the different following approaches; but before trying another method clean the repo `git clean -f -x -d; git reset --hard`

```
dotnet restore
dotnet build
```

Alternatively you can run;
```
dotnet restore
msbuild MissionPlanner.sln /property:Configuration=Release
```

Yet another alternative method I've tried
```
nuget restore
msbuild MissionPlanner.sln /property:Configuration=Release
```

There are errors, of course, which need a lot of sln, csproj, json edits and, most probably, code changes. But, again, I'm not a developer, I just wanted to be make it easier for someone to start.

Raviu8, 2018-07-25:
Update:
In an attempt to avoid the many file name case sensitivity issues. I've move the repo to a separate ntfs volume and mounted it like that;
`lowntfs-3g /ntfs.fs /mnt -o rw,ignore_case`

This helped a lot to reveal underlying errors without "file not found" issues.

meee1, 2018-07-25:
you do realize that a exe compiled on windows will run on Linux with no changes don't you?

Raviu8, 2018-07-25:
@meee1 Yes, But a lot of bugs are existing.
`mono MissionPlanner.exe` you mean.

It's:
1. Very slow.
2. Sluggish.
3. Windows behave strange - vibrating randomly in strange way.
4. On switching tabs - specially when there is active connection it takes a lot to update displayed data and freeze until done.
5. Changing parameters in "Full parameter list" is a real pain. Specially when you've many to change. You have to switch away, then back again every while as the list is vanished. And the search box causes more bugs.
6. No full parameter tree.
7. Sometimes it just crash on establishing connection.
8. Resizing sub-windows - I don't know exactly what it's called - is a big pain. Like when you need to make waypoints part larger in Flight Plan tab.
9. When you click/switch to any button, menu, sub-menu, tab or so on it's initially unresponsive and you've to wait. This is impractical if you are in-flight.
10. Sometimes you've to work much to know where exactly you've to click for the item to interact. Like changing waypoint type in Flight Plan -> waypoints.
11. No user-friendly way to display the full text of options or description for some parameter in Full Parameter list.
12. Click find in Advanced params to filter the output and try to go page down them mouse wheel -> pain.
13. Bars in servo output display strange output, go on/off randomly. Other functions too have the same bug.
14. You have to click on the needed area to have mouse wheel for example working on it. It doesn't work on another area on the same window unless you click there.

I have experienced other bugs but I cannot remember of all them.

Getting it to run with wine with donet stuff is overall better but very bad ugly eye-hurting fonts. Which I've worked a lot to make it better, just made it less worse.

Overall, it's not native. Other native mono-based projects works very fine recently on Linux.

Raviu8, 2018-07-25:
Update2:
Actually, I don't think I can go further.

I found that I've to declare two variables to get things in the right direction and to avoid major failures.
```
export RuntimeIdentifier=$(dotnet --info | grep "RID" | awk '{print $2}')
export FrameworkPathOverride=/usr/lib/mono/4.6.1-api/
```
References:
https://stackoverflow.com/questions/44770702/build-nuget-package-on-linux-that-targets-net-framework
https://github.com/dotnet/netcorecli-fsc/wiki/.NET-Core-SDK-rc4#using-net-framework-as-targets-framework-the-osxunix-build-fails

This allowed me to see "native" errors.

I've created a small patch - below - against master that could get `dotnet restore /p:Configuration=Release` working.
Actually, I know that just changing version numbers this way is not right, but I just want it to go forward and solve downgrade errors.

```
diff --git a/ExtLibs/Comms/MissionPlanner.Comms.csproj b/ExtLibs/Comms/MissionPlanner.Comms.csproj
index 073e439..f6b98f7 100644
--- a/ExtLibs/Comms/MissionPlanner.Comms.csproj
+++ b/ExtLibs/Comms/MissionPlanner.Comms.csproj
@@ -11,7 +11,7 @@
   </PropertyGroup>
 
   <ItemGroup>
-    <PackageReference Include="log4net" Version="2.0.8" />
+    <PackageReference Include="log4net" Version="2.0.0" />
     <PackageReference Include="System.IO.Ports" Version="4.5.0" />
   </ItemGroup>
 
diff --git a/ExtLibs/Controls/MissionPlanner.Controls.csproj b/ExtLibs/Controls/MissionPlanner.Controls.csproj
index c4423bf..61e67c7 100644
--- a/ExtLibs/Controls/MissionPlanner.Controls.csproj
+++ b/ExtLibs/Controls/MissionPlanner.Controls.csproj
@@ -312,7 +312,7 @@
   </ItemGroup>
   <ItemGroup>
     <PackageReference Include="log4net">
-      <Version>2.0.8</Version>
+      <Version>2.0.0</Version>
     </PackageReference>
     <PackageReference Include="OpenTK">
       <Version>3.0.1</Version>
@@ -331,4 +331,4 @@
     </PackageReference>
   </ItemGroup>
   <Import Project="$(MSBuildBinPath)\Microsoft.CSharp.targets" />
-</Project>
\ No newline at end of file
+</Project>
diff --git a/ExtLibs/GDAL/GDAL.NET.csproj b/ExtLibs/GDAL/GDAL.NET.csproj
index e8e067f..6f9470a 100644
--- a/ExtLibs/GDAL/GDAL.NET.csproj
+++ b/ExtLibs/GDAL/GDAL.NET.csproj
@@ -27,7 +27,7 @@
 
   <ItemGroup>
     <PackageReference Include="GDAL" Version="1.11.1" />
-    <PackageReference Include="log4net" Version="2.0.8" />
+    <PackageReference Include="log4net" Version="2.0.0" />
     <PackageReference Include="MP.GDAL.Native" Version="2.2.3" />
     <PackageReference Include="System.Drawing.Common" Version="4.5.0" />
     <ProjectReference Include="..\GMap.NET.Core\GMap.NET.Core.csproj" />
diff --git a/ExtLibs/Mavlink/MAVLink.csproj b/ExtLibs/Mavlink/MAVLink.csproj
index 59844d8..65ca1b6 100644
--- a/ExtLibs/Mavlink/MAVLink.csproj
+++ b/ExtLibs/Mavlink/MAVLink.csproj
@@ -21,7 +21,7 @@
   </ItemGroup>
[... 124 lines cut ...]
--- a/SikRadio/SikRadio.csproj
+++ b/SikRadio/SikRadio.csproj
@@ -238,7 +238,7 @@
   </ItemGroup>
   <ItemGroup>
     <PackageReference Include="log4net">
-      <Version>2.0.8</Version>
+      <Version>2.0.0</Version>
     </PackageReference>
     <PackageReference Include="System.Drawing.Common">
       <Version>4.5.0</Version>
@@ -258,4 +258,4 @@
   <Target Name="AfterBuild">
   </Target>
   -->
-</Project>
\ No newline at end of file
+</Project>
```

One small change is add `<Reference Include="netstandard" />` to MissionPlanner.csproj, This solved many errors as of this issue https://github.com/dotnet/standard/issues/542

Now most of the errors are about types and such programming things :)

Hoping someone can continue.

kikislater, 2018-07-26:
Well resumed @Raviu8 
I will add : 
15 - No GPS information in main screen :  https://github.com/ArduPilot/MissionPlanner/issues/1292
Could be workaround by adding item in HUD

16 - Gauges display problem 
[image] May be related to this : https://github.com/ArduPilot/MissionPlanner/issues/79

17 - Simulation not usable

18 - Auto log analysis crash

Mission Planner constantly use 16% of CPU on i7 4790k which is a lot. On slow computer like intel atom tablet or laptop even with i5 3rd generation, artefacts in panels appears.
I found that it runs better with latest mono version, but still have some problems and slow interface. I use mono 5.12.0.260-1 at this time.

atmadeep, 2018-12-11:
Hi, It seems to me that these problems are persistent. I have Ubuntu 18.04. I tried it with mono (mono-complete). The features which are a "Minus" are essential to my project.  Can someone please answer my following queries:
0. How is 'Mission planner 1.x' different than 'APM 2.x'? The list of options in Mission planner is different than the ones in APM. Can you please provide me a list or guide which I can use as a lookup table?

1. Is there a way to run Mission planner 1.x version 'natively' on ubuntu/Linux? As most of the tutorials which I see, they use Mission planner v/s APM, and would like to use it for my project.

2. I cannot use SITL on ubuntu with APM 2.x. The dependencies are broken/ mismatched between component viz. pymavlink, dronekit-sitl and mavproxy. Can you please provide a solution or a guide which I can use to create mission for my quadcopter for aerial imagery?

kikislater, 2018-12-16:
Why not using qgroundcontrol : https://github.com/mavlink/qgroundcontrol ? No need to build it, there is appimage. And it runs natively instead of MP.

kschoelpple, 2020-08-24:
I'm simply using Wine. MP works straight away and have been using it like that for the past two years. (on 18.04 and now on 20.04)
However, there are some limitations. 
More info:
https://appdb.winehq.org/objectManager.php?sClass=version&iId=38641&iTestingId=107520