ArduPilot/MissionPlanner #1151: Read waypoint file Automatically

MusabAlaqeede, 2016-02-21:
I would like to edit the mission planner to automatically read the text file that has the coordinates once its updated. where should I look ?

magicrub, 2016-02-21:
+1. Perhaps there could be a Check box in general settings to auto read the
waypoints on connect

meee1, 2016-02-21:
there already is a setting to read wp's con connect.
look in the planner settings page, under waypoints.

meee1, 2016-02-21:
@MusabAlaqeede can you tell me a use case? ie an example work flow so i can understand your request.

MusabAlaqeede, 2016-02-22:
@meee1  I have a system that generates a waypoints in a text file on the desktop, I want to make the mission planner to automatically take these waypoints from the text file once it's updated.

meee1, 2016-02-22:
seems like a very specific workflow you are after. and not something i would concider normal use.
was the intent to start missionplanner after the file was created? or have it look at it for changes?

MusabAlaqeede, 2016-02-22:
The system I have created is a mobile application that sends out a gps location to the basestation once the drone is requested by the mobile user. The base station will generate a text file on the desktop that is ready for the mission planner to read from, The issue I have is that I have to manually find and open the text file rather than the mission planner automatically take up the waypoints from the text file once its created or updated.

meee1, 2016-02-22:
ok, and what about uploading them to the autopilot? is the manual?

MusabAlaqeede, 2016-02-22:
Well the plan is that once the "Flight plan" reads the text file and the waypoints appear on the map, the admin will manually write the waypoints onto the drone .... we are assuming that once the administrator receives the request he will need to equip the drone with the order before sending it to the requester. its something like the amazon delivery project

meee1, 2016-02-23:
i think the best method to implment this would be as a mission planner plugin. as i dont think its relivant for the main code.

MusabAlaqeede, 2016-02-23:
Please elaborate

meee1, 2016-02-27:
well, your original request was for where to start looking.

so my suggestions are.
use the mission planner plugin interface, this way you can have it look for your file/in a directory for the file, and then import that into flightplanner, and switch to the flightplanner screen.

your idea is to do a single decrete task, so a plugin i believe would be best suited.

magicrub, 2016-02-28:
I had never used the auto WP load on connect until I learned of it via this
issue. I found it annoying to use due to the pop up dialog boxes. Is there
a way to bypass that for the auto load?

well, your original request was for where to start looking.

so my suggestions are.
use the mission planner plugin interface, this way you can have it look for
your file/in a directory for the file, and then import that into
flightplanner, and switch to the flightplanner screen.

your idea is to do a single decrete task, so a plugin i believe would be
best suited.

meee1, 2016-02-28:
the reason for the prompt is if you plan a mission before you connect, it doesnt wipe the mission from flightplanner. it was done the way it is currently by design.

sevyon, 2016-05-16:
Hi i am also on a similar project but mine has to read and upload the wp to the autopilot automatically.
How can i go about to do this ?

ElefLee, 2017-03-20:
Hello MusabAlaqeede

I am in similar project too and my base station generates a text file on the desktop that is ready for the mission planner to read from. I also look for a script that the mission planner will automatically take up the waypoints from the text file once its created or updated.

How could i get in touch with you?

Thanks a lot

Diziano, 2018-02-26:
@MusabAlaqeede @ElefLee 
I need the same solution, it's for an academic job. Did you find a solution?
Thank you.

RisingPhoenix247, 2018-07-30:
Hello
i need a solution for the same as i've the same problem statement as @sevyon .
@MusabAlaqeede @meee1 Any leads or help will be very helpful.
Thanks in advance.

RisingPhoenix247, 2018-08-07:
namespace MissionPlanner.GCSViews
{
    public partial class FlightPlanner : MyUserControl, IDeactivate, IActivate
    {
        private static readonly ILog log = LogManager.GetLogger(MethodBase.GetCurrentMethod().DeclaringType);
        int selectedrow;
        public bool quickadd;
        bool isonline = true;
        bool sethome;
        bool polygongridmode;
        bool splinemode;
        altmode currentaltmode = altmode.Relative;

        bool grid;

        public static FlightPlanner instance;

        public List<PointLatLngAlt> pointlist { get; set; } = new List<PointLatLngAlt>();

        static public Object thisLock = new Object();
        private ComponentResourceManager rm = new ComponentResourceManager(typeof (FlightPlanner));

        private Dictionary<string, string[]> cmdParamNames = new Dictionary<string, string[]>();

        List<List<Locationwp>> history = new List<List<Locationwp>>();

        List<int> groupmarkers = new List<int>();

        public enum altmode
        {
            Relative = MAVLink.MAV_FRAME.GLOBAL_RELATIVE_ALT,
            Absolute = MAVLink.MAV_FRAME.GLOBAL,
            Terrain = MAVLink.MAV_FRAME.GLOBAL_TERRAIN_ALT
        }
       /* protected void Page_Init(object sender, EventArgs e)
        {
            readQGC110wpfile("C:\\Users\\SAM\\Downloads\\waypoint1.waypoints", true);
        }*/

I'm trying to load the waypoint as soon as the flightplanner page is loaded/opened. This should work logically speaking but it still isn't working. Let me know how it can be modified if in case i'm wrong.

RisingPhoenix247, 2018-08-08:
if you already know which file you want it to read from then here's the solution that worked 💯  for me 
Go to flightplanner.cs
find MainMap_OnTileLoadStart()
Here is the modifications that are to be made :

 void MainMap_OnTileLoadStart()
        {
            readQGC110wpfile("C:\\Users\\SAM\\Downloads\\waypoint1.waypoints", true);
        }

Comment the other code which is already written for this function.
save it and load mission planner and click on flightPlanner. You should now be able to see the waypoints plotted on the map automatically!! BOOM!
Cheers 🍻

meee1, 2018-08-08:
add your code to the function Activate

closed 2018-08-08 by meee1