ArduPilot/MissionPlanner #1152: How to use COMMAND_INT

m11s, 2016-02-22:
From what I understand COMMAND_INT is in Mission Planner's Mavlink message definations [here](https://github.com/diydrones/MissionPlanner/blob/fea7ba06b4c1f6e6d1def8481153ac4941d1c321/ExtLibs/Mavlink/generator/message_definitions/common.xml#L2326) and is said to be supported in autopilot [here](https://github.com/diydrones/MissionPlanner/blob/fea7ba06b4c1f6e6d1def8481153ac4941d1c321/ExtLibs/Mavlink/generator/message_definitions/common.xml#L2326). 
Mission Planner uses COMMAND_LONG by default. 
It was said in a Mavlink issue [here](https://github.com/mavlink/mavlink/issues/426#issuecomment-132340142) that COMMAND_LONG and COMMAND_INT were interchangable.

Is there any way to use COMMAND_INT in Mission Planner?

meee1, 2016-02-22:
command_int it not used anywhere in MP either.

i should first ask what is your intent? neither of these commands are used to send waypoints to the autopilot, they are used to send commands to the autopilot.
they use mission_item

m11s, 2016-02-22:
My intent was to use COMMAND_INT to achieve cm accurate waypoints as mentioned in the link below.
https://github.com/diydrones/ardupilot/issues/1907

meee1, 2016-02-23:
MISSION_ITEM_INT is the method it should be done using. this would require changes to the flight code however.

m11s, 2016-02-23:
What changes would it take? That you know of. [This](https://github.com/diydrones/ardupilot/blob/27ed9e15616196f89b8ae180e4964e5fdfc09ef1/libraries/GCS_MAVLink/MAVLink_routing.cpp#L420) is the only reference I found to it in APM.

That's why I initially thought of COMMAND_INT because it is already partially in APM [here](https://github.com/diydrones/ardupilot/blob/e09e9a313ecb5e78046093c537421948230b0911/ArduCopter/GCS_Mavlink.cpp#L1155) and just needs more MAV_CMDs added like [COMMAND_LONG](https://github.com/diydrones/ardupilot/blob/e09e9a313ecb5e78046093c537421948230b0911/ArduCopter/GCS_Mavlink.cpp#L1193) has.

meee1, 2016-02-27:
none of what you linked is adding support for the message.
ive started to add support
https://github.com/meee1/ardupilot/tree/mission_int2
but am not happy with the solution yet.

m11s, 2016-03-02:
Ok. About how long do you think it will be until the betas are ready to test fly?

meee1, 2016-03-02:
depends on when the change gets pulled in.
Mission planner beta is ready
https://github.com/diydrones/ardupilot/pull/3688

m11s, 2016-03-03:
They can't get pulled in yet because the Travis CI build failed on "sitltest". 
Thanks for the MP updates!