ArduPilot/MissionPlanner #1155: HUD user items stop responding after long missions.

GrahamDyer, 2016-02-24:
We fly +/-2 hour missions and if we reconnect to the plane to start the second or third mission the user items on the HUD stop updating requiring a MP restart to get everything working again. This started with v1.3.34.

meee1, 2016-02-24:
is it the same plane every time?

GrahamDyer, 2016-02-24:
Yes, the planes' batteries get changed so Pixhawk reboots but it's normally the same plane. Upon reconnecting the HUD user items don't update, staying either on 0 or whatever they were last on when disconnected from the plane previously.

magicrub, 2016-02-24:
Check that the sysid params are the same on the aircraft. Also, how are you
connecting via?

GrahamDyer, 2016-02-24:
Via RFD900's radio's.

iskess, 2016-02-24:
I have noticed the same problem. The values freeze when reconnected to the same aircraft. I have to de-select and re-select the user item to get it to work again. I did this a dozen times when trying to troubleshoot the Throttle% for reverse thrust, but forgot to report it. 
(PS. I think throttle% should always be on the HUD, not an optional user item, IMHO)

meee1, 2016-02-25:
please try the latest beta, i made a mod last night

meee1, 2016-02-27:
change was here https://github.com/diydrones/MissionPlanner/commit/7e5eae50d85a18185fec863308803fe48d48f043

closed 2016-02-27 by meee1

iskess, 2016-04-06:
This is still not fixed. The HUD user items were frozen during my whole flight today. I can duplicate while playing back the tlog, by pressing pause, then resuming playback.

meee1, 2016-04-06:
I might get you to test a change to verify this one.
the tlog playback issue is related to the pausing reset, so not realy a solid link/repeatable.