ArduPilot/MissionPlanner #1159: I can't complile Mission Planner

sgh055, 2016-02-26:
I download Mission Planner [Mission Planner](http://ardupilot.com/downloads/?did=82) here and [Mission Planner(github)](https://github.com/diydrones/MissionPlanner) here

1. installed Mission Plnnaer in my PC(C:\Program Files (x86)\Mission Planner)

2. Open this Visual studio 2013 ultimate[image]

3. Add Reference Path [image]

4. Compiled -> Error

5.  So I copy .dll files at folder(ExtLibs\Mavlink\bin\Debug, ExtLibs\Comms\bin\Debug, ExtLibs\Controls\bin\Debug, ExtLibs\Utilities\bin\Debug)[image]

6. finaly, I can see so many~~ errors :(

7. How I fix this problems

I following here(http://dev.ardupilot.com/wiki/buildin-mission-planner/) what i missing?

SkiTiSu, 2016-02-26:
Could you please copy the error details?

I guess that because you have not configured the python environment.

Since commit a396b5a61697bd8c9e0e12c98d9e08798f7533a5, the Mavlink library need be generated on every build. Make sure you have installed python and set up the environment variable.

sgh055, 2016-02-26:
@SkiTiSu Hi, I installed python 2.7(https://www.python.org/downloads/) but errors retain

here my errors 

```
Error1	The command "rd /s /q "D:\Download\MissionPlanner\ExtLibs\Mavlink\generator\Csharp"
python D:\Download\MissionPlanner\ExtLibs\Mavlink\generator\mavgen.py --lang=csharp -o "D:\Download\MissionPlanner\ExtLibs\Mavlink\generator\Csharp" --wire-protocol=1.0 "D:\Download\MissionPlanner\ExtLibs\Mavlink\generator\message_definitions\ardupilotmega.xml"  
copy /y "D:\Download\MissionPlanner\ExtLibs\Mavlink\generator\Csharp\ardupilotmega\mavlink.cs" "D:\Download\MissionPlanner\ExtLibs\Mavlink\Mavlink.cs"
" exited with code 1.	MAVLink
Error	15	'MAVLink.MAVLinkParam' does not contain a definition for 'TypeAP' and no extension method 'TypeAP' accepting a first argument of type 'MAVLink.MAVLinkParam' could be found (are you missing a using directive or an assembly reference?)	D:\Download\MissionPlanner\Controls\MavlinkCheckBoxBitMask.cs	94	45	MissionPlanner
Error	22	'MAVLink.MAVLinkParam' does not contain a constructor that takes 4 arguments	D:\Download\MissionPlanner\Mavlink\MAVLinkInterface.cs	796	45	MissionPlanner
Error	23	'MAVLink.MAVLinkParam' does not contain a constructor that takes 4 arguments	D:\Download\MissionPlanner\Mavlink\MAVLinkInterface.cs	990	53	MissionPlanner
Error	24	'MAVLink.MAVLinkParam' does not contain a constructor that takes 4 arguments	D:\Download\MissionPlanner\Mavlink\MAVLinkInterface.cs	1154	45	MissionPlanner
Error	27	'MAVLink.MAV_SYS_STATUS_SENSOR' does not contain a definition for 'MAV_SYS_STATUS_REVERSE_MOTOR'	D:\Download\MissionPlanner\CurrentState.cs	2243	102	MissionPlanner
Error	28	'MAVLink.MAV_SYS_STATUS_SENSOR' does not contain a definition for 'MAV_SYS_STATUS_REVERSE_MOTOR'	D:\Download\MissionPlanner\CurrentState.cs	2244	95	MissionPlanner
Error	29	'MAVLink.MAVLinkParam' does not contain a constructor that takes 4 arguments	D:\Download\MissionPlanner\Mavlink\MAVLinkInterface.cs	3089	56	MissionPlanner
Error	32	Metadata file 'D:\Download\MissionPlanner\bin\Debug\MissionPlanner.exe' could not be found	D:\Download\MissionPlanner\ExtLibs\Grid\CSC	MissionPlanner.Grid
Error	33	Metadata file 'D:\Download\MissionPlanner\bin\Debug\MissionPlanner.exe' could not be found	D:\Download\MissionPlanner\ExtLibs\MissionPlanner.Stats\CSC	MissionPlanner.Stats
Error	34	Metadata file 'D:\Download\MissionPlanner\bin\Debug\MissionPlanner.exe' could not be found	D:\Download\MissionPlanner\ExtLibs\TrackerHome\CSC	TrackerHomeGPS
Error	35	Metadata file 'D:\Download\MissionPlanner\bin\Debug\MissionPlanner.exe' could not be found	D:\Download\MissionPlanner\ExtLibs\SimpleGrid\CSC	MissionPlanner.SimpleGrid
Error	36	Metadata file 'D:\Download\MissionPlanner\bin\Debug\MissionPlanner.exe' could not be found	D:\Download\MissionPlanner\ExtLibs\MissionPlanner.Gridv2\CSC	MissionPlanner.Gridv2
Error	37	Metadata file 'D:\Download\MissionPlanner\bin\Debug\MissionPlanner.exe' could not be found	D:\Download\MissionPlanner\ExtLibs\ExtGuided\CSC	ExtGuided

```

SkiTiSu, 2016-02-26:
@sgh055 the python returns 1(error). You can run
`python D:\Download\MissionPlanner\ExtLibs\Mavlink\generator\mavgen.py --lang=csharp -o "D:\Download\MissionPlanner\ExtLibs\Mavlink\generator\Csharp" --wire-protocol=1.0 "D:\Download\MissionPlanner\ExtLibs\Mavlink\generator\message_definitions\ardupilotmega.xml"`
manually in CMD and find out the reason.
The errors below are all caused by this.

sgh055, 2016-02-26:
@SkiTiSu 
can you tell me solution this error??
because i already your solution..but i can't...
maybe could you show me about your solution??? video or manual???

SkiTiSu, 2016-02-26:
@sgh055 I'm sorry, I didn't get this error before. You can run that command manually in CMD, then you will know the details about this error.

sgh055, 2016-02-26:
@SkiTiSu  Thanks! Solved my problem!!
The reason is Python environment variable setting

closed 2016-02-27 by meee1