ArduPilot/MissionPlanner #1162: Latest fails to build with Visual Studio

kjeremy, 2016-03-01:
I'm seeing a lot of build failures on `master` today with Visual Studio.

```
8>------ Build started: Project: SharpKml, Configuration: Debug Any CPU ------
3>C:\Program Files (x86)\MSBuild\14.0\bin\Microsoft.Common.CurrentVersion.targets(1249,5): error MSB3073: The command "rd /s /q "C:\Projects\MissionPlanner\ExtLibs\Mavlink\generator\Csharp"
3>C:\Program Files (x86)\MSBuild\14.0\bin\Microsoft.Common.CurrentVersion.targets(1249,5): error MSB3073: python C:\Projects\MissionPlanner\ExtLibs\Mavlink\generator\mavgen.py --lang=csharp -o "C:\Projects\MissionPlanner\ExtLibs\Mavlink\generator\Csharp" --wire-protocol=1.0 "C:\Projects\MissionPlanner\ExtLibs\Mavlink\generator\message_definitions\ardupilotmega.xml"  
3>C:\Program Files (x86)\MSBuild\14.0\bin\Microsoft.Common.CurrentVersion.targets(1249,5): error MSB3073: copy /y "C:\Projects\MissionPlanner\ExtLibs\Mavlink\generator\Csharp\ardupilotmega\mavlink.cs" "C:\Projects\MissionPlanner\ExtLibs\Mavlink\Mavlink.cs"
3>C:\Program Files (x86)\MSBuild\14.0\bin\Microsoft.Common.CurrentVersion.targets(1249,5): error MSB3073: " exited with code 1.
```
...
```
28>------ Build started: Project: ExtGuided, Configuration: Debug Any CPU ------
24>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\ExtLibs\Mavlink\bin\Debug\MAVLink.dll' could not be found
24>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\ExtLibs\Comms\bin\Debug\MissionPlanner.Comms.dll' could not be found
24>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\ExtLibs\Controls\bin\Debug\MissionPlanner.Controls.dll' could not be found
24>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\bin\Debug\MissionPlanner.exe' could not be found
24>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\ExtLibs\Utilities\bin\Debug\MissionPlanner.Utilities.dll' could not be found
25>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\ExtLibs\Mavlink\bin\Debug\MAVLink.dll' could not be found
25>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\ExtLibs\Comms\bin\Debug\MissionPlanner.Comms.dll' could not be found
25>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\ExtLibs\Controls\bin\Debug\MissionPlanner.Controls.dll' could not be found
25>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\bin\Debug\MissionPlanner.exe' could not be found
25>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\ExtLibs\Utilities\bin\Debug\MissionPlanner.Utilities.dll' could not be found
28>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\ExtLibs\Mavlink\bin\Debug\MAVLink.dll' could not be found
28>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\ExtLibs\Comms\bin\Debug\MissionPlanner.Comms.dll' could not be found
28>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\ExtLibs\Controls\bin\Debug\MissionPlanner.Controls.dll' could not be found
28>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\bin\Debug\MissionPlanner.exe' could not be found
28>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\ExtLibs\Utilities\bin\Debug\MissionPlanner.Utilities.dll' could not be found
27>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\ExtLibs\Mavlink\bin\Debug\MAVLink.dll' could not be found
27>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\ExtLibs\Comms\bin\Debug\MissionPlanner.Comms.dll' could not be found
27>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\ExtLibs\Controls\bin\Debug\MissionPlanner.Controls.dll' could not be found
27>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\bin\Debug\MissionPlanner.exe' could not be found
27>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\ExtLibs\Utilities\bin\Debug\MissionPlanner.Utilities.dll' could not be found
23>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\ExtLibs\Mavlink\bin\Debug\MAVLink.dll' could not be found
23>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\ExtLibs\Comms\bin\Debug\MissionPlanner.Comms.dll' could not be found
23>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\ExtLibs\Controls\bin\Debug\MissionPlanner.Controls.dll' could not be found
23>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\bin\Debug\MissionPlanner.exe' could not be found
23>CSC : error CS0006: Metadata file 'C:\Projects\MissionPlanner\ExtLibs\Utilities\bin\Debug\MissionPlanner.Utilities.dll' could not be found
```

meee1, 2016-03-01:
you need to have python install on the build pc, and in your path.

kjeremy, 2016-03-01:
This is a new requirement?

meee1, 2016-03-01:
yes, only new

kjeremy, 2016-03-02:
After installing python I still get:

```
error MSB3073: The command "rd /s /q "C:\Projects\MissionPlanner\ExtLibs\Mavlink\generator\Csharp"
```

I'm not sure where 'rd' comes from.

meee1, 2016-03-08:
solved by pull request

closed 2016-03-08 by meee1