ArduPilot/MissionPlanner #1170: Various Issues On Mission Planner 1.3.36

Fortesque0, 2016-03-08:
Hello:
I found some issues on this versión.

-TRADITIONAL HELI: - On heli setup Window, Rotor Speed Control has not ítems (on SITL too)
[image]
  -Failsafe Settings: The failsafe it´s looking for throttle on channel 3, but the throtle input it´s on Channel 8 for traditional helicopters.

  -Throttle cannel setting: Using a ppm converter, i can change the cannel 3 to 8, crossing receivers outputs, but using a SBUS or PPM receiver, ¿How we do channel change?
 

- SITL:  -When the joystick it´s enabled, Mission Planner crash
[image]

 - SIM Link Star/Stop always sent Xplane setting (with other simulator selected)
[image]

-AIRPLANE: - The Arm Checks has not selectable items
[image]

Thanks in advance!

magicrub, 2016-03-08:
please make separate issues for separate problems.

Fortesque0, 2016-03-08:
Hello magicrub,
I will do.
Thanks

Fortesque0, 2016-03-16:
Hello,

Could anyone take a look to this issue?
Thanks

meee1, 2016-03-16:
please update to latest mp versions.
then goto help > update. to refresh the param list.

Fortesque0, 2016-03-16:
I did  it yesterday exactly like do you say. From MP 1.35 (Working parameter), to MP 1.37. Star SITL for Helicopter and parammeter doesn´t appears.
I will do today erasing MP folder and downloading Last MP .MSI

meee1, 2016-03-16:
after installing 1.37, you need to do another update to refresh the param list.

Fortesque0, 2016-03-17:
Thanks. I did erasing folder and reinstalling. 
I didn't know ctrl-f option. I have been looking on MP wiki about this option, and i didn't find información.
I think its very important to document all changes-options. One more time, i offer myself to help updating the documentación.
I close this issue.
Thanks.

closed 2016-03-17 by Fortesque0