ArduPilot/MissionPlanner #1172: Various Issues On Mission Planner 1.3.36 Using Joystick on SITL , Mission Planner Crash

Fortesque0, 2016-03-08:
Hello:

When you use SITL and then you change joystick to enabled, Mission Planner crash.

[image] 

Thanks in advance!

magicrub, 2016-03-08:
Do you have a physical joystick connected? What kind? Is this SITL on windows?

meee1, 2016-03-08:
what type/brand of joystick are you using?

Fortesque0, 2016-03-08:
Hello:

The same joystick that i was using on previous misión planner versión without problem, NGS Black Hawk via USB.
Windows and mission planner recognize it as
[image]

Fortesque0, 2016-03-08:
And i use all SITL options  throught missión planner on Windows 7

meee1, 2016-03-08:
what was the last version that worked? the joystick code hasn't changed in 4 months

Fortesque0, 2016-03-08:
On MP 1.3.35 Works fine. I will try to unisnstal and install joystick again.

Fortesque0, 2016-03-08:
I have unninstall and install again joystick and issue persist. I will reinstal MP1.3.35 and take a shootscreen using SITL and Joystick.

Fortesque0, 2016-03-09:
Hello,
I have reinstall MP 1.35 and joystick Works fine. I have unninstall MP1.35, delete folders and restart computer. Install MP1.36, configure and safe joystick, and again, MP crashes on SITL when Joystick its selected.

meee1, 2016-03-14:
done

closed 2016-03-14 by meee1

Fortesque0, 2016-03-14:
Thanks Michael, now Works.