ArduPilot/MissionPlanner #1174: Various Issues on Mission Planer ?.?.? Traditional Helicopter Failsafe Settings

Fortesque0, 2016-03-08:
Hello,
 I do not know exactly which version stopped working or failed, but in the current and previos versión, this happends.
The failsafe it´s looking for throttle on channel 3, but the throtle input it´s on Channel 8 for traditional helicopters. So the failsafe doesn´t work properly. 
And how will work failsafe throttle ouput, looking for diferents RSC Throttle inputs (Ch8, Dissable or setpoint)??

Thanks in advance

meee1, 2016-03-08:
is this a Mission planner issue, or a firmware issue? ie what screen I'm mp is wrong?

magicrub, 2016-03-08:
doesn't sound like a MP problem to me

Fortesque0, 2016-03-08:
I don´t know if is a MP or Firmware problema. Who set´s the throttle channel? I am taking shot screens, one moment

Fortesque0, 2016-03-08:
You can see that throttle (cannel 8),its under failsafe set point, but pixhak its not on failsade mode. Channel 3 (Collective pitch on helicopter) its over failsafe point too.
[image]
Now, you can see that cannel 3 its under failsafe setpoint, and pixhawk goes to failsafe mode
[image]

The same happends if you are using Helicopter sitl . Firmware or MP, i don´t know where it´s the issue.

Fortesque0, 2016-03-09:
The SITL Helicopter doesn't take off to. Could be this relationated? Cyclic servos output increases, but the heli doesn't moves.

rmackay9, 2016-03-10:
The failsafe is triggered from channel 3.  So the "Radio 3" needs to go below 1045.  It also won't trigger while the vehicle is disarmed.
This is really a support question about the trad heli firmware not mission planner so doesn't really belong in github (which is for bugs and enhancement requests) so closing.

closed 2016-03-10 by rmackay9