ArduPilot/MissionPlanner #1176: Virus detected downloading Mission Planner SITL firmwares

Fortesque0, 2016-03-08:
Hello:
Avast antivirus detects Drep virus when SITL ask for firmwares.
[image]

meee1, 2016-03-09:
DomainRep is a new feature of Avast, so let me explain a bit. It blocks EXE files downloads if these conditions are *all* met:
1. The file is not prevalent enough, ie. not enough Avast users launched the file yet,
2. The domain is not prevalent enough, ie. not enough Avast users downloaded (any) EXE files from the domain yet,
3. The file is not signed or Avast does not trust the signature.

Once one of these conditions are not met anymore, Avast will stop flagging the download. In other words, just wait until more people try to download the file, or digitally sign your files :-).

from https://forum.avast.com/index.php?topic=163221.0

meee1, 2016-03-09:
it is safe to ignore.

closed 2016-03-09 by meee1

Fortesque0, 2016-03-09:
Thanks Michael