ArduPilot/MissionPlanner #1177: Mission Planner SITL doesn´t star JBSIM graphical simulator

Fortesque0, 2016-03-08:
Hello:
I change options, but allways starts xplane link
[image]
Thanks in advance

meee1, 2016-03-09:
that is a display issue only. it will say that on all sim types.

Fortesque0, 2016-03-09:
Hello Michael:

I can´t make JSBSim to work (For have a graphical 3d simulated copter/plane). I am looking on to console and there is an error report
[image]
Could this be the reasson?
Thanks

meee1, 2016-03-09:
need to take a step back. jsbsim in not a 3d simulator, it is a physics engine. what is it your expecting

Fortesque0, 2016-03-09:
Sorry, i was confused. The simulator that was integrated on previous versions was Flight Gear, so, why doens´t starts if i have selected , and why i can select  jsbsim (i suposse that jsbsim its the engine for sitl)
Thanks

Fortesque0, 2016-03-14:
Finally I managed to run the Flightgear with the SITL. Honestly, it is not collected anywhere to properly configure the most current version.
The manuals in Wiki are outdated and useless to make changes if not properly documented in the Wiki.
I offer volunteer to help update the wiki. That any project manager contact me if you are interested in my offer.
 regards

Fortesque0, 2016-03-28:
Could be possible to write correct instructions on wiki for 3d simulator running on windows?

Fortesque0, 2016-08-11:
Hello,
Any updates for this?

closed 2017-07-13 by Fortesque0