ArduPilot/MissionPlanner #1181: Survey Grid: CAM_TRIGG_DIST not added to beginning of mission

jmachuca77, 2016-03-10:
When creating a survey grid and selecting cam_trigg_dist as the cam triggering method, the message is not added to the mission. 
[image]
[image]

meee1, 2016-03-10:
Jamie, your correct, turn on breakup starts as a workaround.

ReelAir, 2016-03-10:
Cheers Jaime for posting this for me... And thank you Michael for the work around

meee1, 2016-03-10:
I will push a new beta now, with the fix.

ReelAir, 2016-03-10:
Cheers Michael.. Your a champ...

Tex
0418102767

Owner/Pilot/Chief Controller

CASA.UOC.0124 

tex@reelairimagery.com.au <mailto:tex@reelairimagery.com.au> 

 <https://www.facebook.com/ReelAirImagery?ref=tn_tnmn>

+-------------------------- Dyslexic / Dysgraphic  Email Disclaimer -------------------+

I'm Dyslexic / Dysgraphic. Reasonable care has been taken to ensure this
email is legible. However I do not take responsibility for offending you with 
poor spelling or grammar.
 <https://www.facebook.com/ReelAirImagery?ref=tn_tnmn>
 <https://www.facebook.com/ReelAirImagery?ref=tn_tnmn> <https://www.facebook.com/ReelAirImagery?ref=tn_tnmn>http://en.wikipedia.org/wiki/Dysgraphia <http://en.wikipedia.org/wiki/Dysgraphia>        http://www.dyslexia.com/famous.htm <http://www.dyslexia.com/famous.htm>

closed 2016-03-10 by meee1

jmachuca77, 2016-03-12:
Working on the latest beta. Break up starts -> this is for not taking pictures during turns?

meee1, 2016-03-12:
@jmachuca77 correct, trig dist is turned off and on at the start and end of each line